cta-observatory/ctapipe #660: DataContainer.dl2 should be `Map[tel_id: List[DL2Container]]`

MaxNoe, 2018-02-22:
So that a generic algorithm can do something like this:

```
event.dl2[current_telescope].append(hillas)
event.dl2[current_telescope].append(my_fancy_reconstruction_result)
```

And a DL2 writer could simply loop over all the DL2 containers and the necessary metadata

kosack, 2018-03-02:
Would that work with writing out to a table with a `TableWriter`?  Having a generic list makes the concept of Containers being mappings to database tables a bit more complicated, but it could be done (actually it may already work, with your last patch to be able to write a list of containers). 

We might need to add some way to append a prefix to a column name in the TableWriter, however, so that if you have multiple algorithms that produce e.g.`energy` columns, you'd get `impact_energy`, `hillas_energy` in the output table.  That might imply using a Map instead of List so that you map *algorithm_id* -> *container of values*, and then one could just prepend the *algorithm_id*.   Then you end up with something like `dl1.tel[x][algorithm]`   

The original concept I had in mind was not so far different, just inverted: I had assumed we'd have e.g. `dl1.tel[x].hillas_params[algorithm_name], or `dl1.tel[x].image[algorithm]`, etc., but it's true just having a list of generic containers is more flexible.

MaxNoe, 2018-03-02:
Container could simply have a `prefix` member that could default to `container.__class__.__name__`

maxnoe, 2021-12-06:
We decided for `Map[Str, Result]`

closed 2021-12-06 by maxnoe