cta-observatory/ctapipe #675: importorskip considered harmful?

dneise, 2018-03-12:
`pytest.importorskip` is routinely used in our tests to skip tests which can only pass when some external dependency is really installed.

This can lead to a whole set of tests being skipped, when there is a single typo in this line:

    pytest.importorskip('external_dependency_name')

This skip does not even show the tests being skipped in yellow in the verbose output, they are skipped entirely. See the screenshot below .. I deliberately added a typo in `test_sst1meventsource.py` in 

    pytest.importorskip('protozfits_typo!!!!')

which resulted in all tests from that file being skipped without any notice in the output, there should be something between the two highlighted lines.

[image]

I am bringing up this topic, because I wanted to ask if supporting this kind of optional dependencies might do more harm than good. What would happen if we'd simply say: This-and-that library is a dependency. Just like numpy is a dependency.

Wouldn't that make our lives and the lives of our colleagues much much easier? And possibly result in a more stable product?

watsonjj, 2018-03-12:
I agree this is bad. For some reason I remember seeing it explicitly informing when these tests are skipped, but I can't find anything that backs up my memory. It seems strange for them to be omitted entirely....

The `pytest -rs` option does add a summary of the reasons why certain tests were skipped...but isn't a full solution.

kosack, 2018-03-12:
I know `@pytest.mark.skipped` will still print them, but say they are skipped (as in your screenshot), but I'm not sure the best way to do  tests optionally... 

Certainly in the continuous integration test suite, we can just ensure all necessary libraries are installed, so it only really affects users running "pytest" locally.

kosack, 2018-03-12:
we could also use [custom markers](https://docs.pytest.org/en/latest/example/markers.html), but I don't really like that either, since if the user doesn't enable the marker, the test is skipped.

kosack, 2018-03-12:
This is what gammapy does: https://github.com/astropy/astropy/issues/5543#issuecomment-265434710

May be nicer, since it would say the test is skipped

dneise, 2018-03-13:
Just so I get this right. 

At the moment we use `importorskip` so that end-users who do not have some of the dependencies installed can execute the tests.
https://github.com/cta-observatory/ctapipe/search?utf8=%E2%9C%93&q=importorskip&type=

We ... in our roles of ctapipe devs would always like to execute all of the tests, and it is dangerous when tests are skipped without us noticing.

Carefully reading the travis output with a human eye to spot skipped tests, feels like a contradiction to the whole continuous integration idea.  

What if we would use custom markers, but by default execute all the tests. An end-user who has not all external dependencies installed will thus see failing tests. Some documentation could tell them how to skip some tests, if the know the do not have the external dependencies installed. 

This way:
 * users are made aware that they do not have all dependencies installed
 * we can be sure all tests are always executed.

Alternatively we could provide different pytest-ini files .. and tell users to do:

    pytest -c no_external_deps.ini 

while we do simply: `pytest`.

MaxNoe, 2018-03-13:
Or maybe do it the other way around? Add an option to execute all tests that is used on travis?

like this maybe:

```python
if os.getenv('FORCE_ALL_TESTS'):
    import external
else:
    pytest.importorskip('external_dependency_name')
```

watsonjj, 2018-03-13:
I personally like the gammapy approach. Its simple, and it ensures that if the test is skipped, then it is clearly shown as skipped in the output summary.

MaxNoe, 2018-03-13:
But we don't want skipped tests on travis, do we?

watsonjj, 2018-03-13:
True, we don't. How about something similar to both the gammapy approach and with your suggestion: if FORCE_ALL_TESTS has been defined in the environment then the decorator does not skip the test.

Then at least the its more clear on developers machines that some tests have skipped because of a missing dependency. I don't like that the skipped tests are hidden at all.

kosack, 2018-03-13:
> But we don't want skipped tests on travis, do we?
 
I don't mind seeing skipped tests, as long as we know why. I think with pytest you can supply a reason for skipping that can be optionally printed.   On travis we can strive to make sure all dependencies are there, hence no skipping.  Right now a few are skipped since they require some input files that are not yet available (like the `ImPACTReconstructor` needs a model table), but those can be resolved in time.

mackaiver, 2018-12-11:
We want to go with something like 

```py3
if os.getenv('FORCE_ALL_TESTS'):
    import external
else:
    external = pytest.importorskip('external_dependency_name')
```
or 

`ctapipe.import_or_skip`

to force running all tests on travis.

closed 2018-12-11 by mackaiver