cta-observatory/ctapipe #676: calib_scale needs citation in HESSIOR1Calibrator

kosack, 2018-03-12:
The `calib_scale` value is the factor needed to transform from mean p.e. units to units of the single-p.e. peak for MC simulations.  In the code it is not documented where the current value came from.

Right now I see in `HESSIOR1Calibrator`:
```py3
calib_scale = 1.05
```

Does anyone know where this value comes from?  For example if I look in `read_hess` (`reconstruct.c` in hessioxxx), the value is 
```
#define CALIB_SCALE 0.92
```

So it's not clear to me which one is correct.

watsonjj, 2018-03-13:
0.92 is the correct one for HESS, 1.05 is the correct one for GCT. This needs to be changed into a lookup based on telescope type.

dneise, 2018-03-13:
Do you think it would help, if this lookup table would also contain (maybe as a comment on top) where these numbers came from?

GernotMaier, 2018-03-19:
0.92 is the factor to transform from p.e. units to units of the single-p.e. peak (assuming the HESS pulse shape for a single pe pulse)

kosack, 2018-03-21:
Ok, so we need to make a lookup for this. I will work on that - I just did the same thing for the gain-switch thresholds. 

Would a table like this suffice? Or do we need more info? 

| cam_id | calib_scale |
|--------| ----------|
| CHEC | 1.05 |
| HESS | 0.92 |
| ...       |          |

kosack, 2018-03-21:
I guess I could also add *measurement_date* and  *comment* fields. That may be good for all lookups, in fact.

kosack, 2018-03-21:
That also means that all the results we have shown so far have the wrong CALIB_SCALE, and so probably affects the sensitivity.  Although now that we have optimized the cleaning thresholds using the wrong scale, it probably doesn't matter much, but makes it harder to compare to MARS/EventDisplay.

kosack, 2018-03-21:
Do we have this value for all cameras somewhere? Or how should we compute it?

GernotMaier, 2018-03-22:
The most consistent way would be to take the actual pulse shape, and calculate these values on the fly. We have all the pulse shapes in the MC (note that they might change with the current process on MC model updates).

I do think think this calibration value depends on the trace integration algorithm used: if you algorithm recovers the mean charge, than no correction is needed. The 0.92 are introduced as most trace algorithms are recovering the ADC counts around the pulse maximum.

Does it actually matter? Only if you really need to recover your image brightness in units of p.e.

For reference, this is how to get the pulseshapes used in the MC model:

1. find in the par-Values-*.yml files in https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/configReports the file name for the file with the pulse shape used (e.g. for MST-FlashCam, this is pulse_FlashCam_raw2.dat
2. look for this file in the directory https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/datFiles (e.g. https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/datFiles/pulse_FlashCam_raw2.dat)

(not that FlashCam is a 'special' example, as an additional pulse-shaping algorithm is applied which is currently not simulated with sim_telarray (MARS and Eventdisplay are doing this on the fly)

All above will hopefully find at some point its way into a data base.

moralejo, 2018-03-22:
In the MARS-based CTA analysis we just applied the factor 0.92 for *all*
cameras - we did not care much, since this has no effect in any performance
parameter:  the analysis algorithms are trained and applied on two MC
samples treated in exactly the same way. So only for comparison of signal
distributions (or e.g. *Size* values) among various analyses it is relevant.

I am not sure I understand what this parameter corrects for:

>
> The most consistent way would be to take the actual pulse shape, and
> calculate these values on the fly. We have all the pulse shapes in the MC
> (note that they might change with the current process on MC model updates).

I do think think this calibration value depends on the trace integration
> algorithm used: if you algorithm recovers the mean charge, than no
> correction is needed. The 0.92 are introduced as most trace algorithms are
> recovering the ADC counts around the pulse maximum.

​If the same trace integration algorithm is used in the reconstruction of
Cherenkov images and in the evaluation of the single photoelectron peak​,
the correction should not depend on the single photoelectron pulse *shape*.
From what I read in the simtelarray comments it depends "on the collection
efficiency, the asymmetry of the single p.e. *amplitude distribution* and
the electronic noise added to the signals". Actually, according to Konrad
the collection efficiency in the simulation is "included" in the single
p.e. amplitude distribution (i.e. the distribution has non-zero value at 0
amplitude). But then, I do not understand what is calib_scale correcting
for - naïvely I would expect it is for the average reduction of integrated
charge (relative to the single-pe case) due to time distribution of photons
in Cherenkov pulses, but then it should always be larger than 1...

Does it actually matter? Only if you really need to recover your image
> brightness in units of p.e.
>
> For reference, this is how to get the pulseshapes used in the MC model:
>
>    1. find in the par-Values-*.yml files in https://forge.in2p3.fr/
>    projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/
>    configReports
>    <https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/configReports>
>    the file name for the file with the pulse shape used (e.g. for
>    MST-FlashCam, this is pulse_FlashCam_raw2.dat
>    2. look for this file in the directory https://forge.in2p3.fr/
>    projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/
>    datFiles
>    <https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/datFiles>
>    (e.g. https://forge.in2p3.fr/projects/cta/repository/show/
>    ASWG/Simulations/MCModelDescription/trunk/datFiles/pulse_FlashCam_raw2.
>    dat
>    <https://forge.in2p3.fr/projects/cta/repository/show/ASWG/Simulations/MCModelDescription/trunk/datFiles/pulse_FlashCam_raw2.dat>
>    )
>
> (not that FlashCam is a 'special' example, as an additional pulse-shaping
> algorithm is applied which is currently not simulated with sim_telarray
> (MARS and Eventdisplay are doing this on the fly)
>
> All above will hopefully find at some point its way into a data base.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

MaxNoe, 2018-03-22:
> Does it actually matter? Only if you really need to recover your image

And of course as soon as we have real data. Because then having the correct number of photons in the MC and not just something proportional to it suddenly becomes important, right?

moralejo, 2018-03-22:
On 22 March 2018 at 15:46, Maximilian Nöthe <notifications@github.com>
wrote:

> Does it actually matter? Only if you really need to recover your image
>
> And of course as soon as we have real data. Because then having the
> correct number of photons in the MC and not just something proportional to
> it suddenly becomes important, right?
>
​If the MC is an accurate representation of the data, it would still be an
overall calibration factor. Same as using "dozens of photoelectrons"
instead of photoelectrons. It would not affect e.g. reconstructed energy,
or anything else.​ And for cross-calibration of telescopes we will have to
go through the MC somehow, e.g. comparing telescope-wise reconstructed
energies for the same events is the easiest way of doing such
cross-calibration.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

-- 
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

MaxNoe, 2018-03-22:
Maybe I'm missing something, but this factor here is only applied to data read from hessio. So if it does not fit, mcs get scaled with respect to data, so you get data/mc mismatch for the energy reconstruction.

moralejo, 2018-03-22:
On 22 March 2018 at 16:03, Maximilian Nöthe <notifications@github.com>
wrote:

> Maybe I'm missing something, but this factor here is only applied to data
> read from hessio. So if it does not fit, mcs get scaled with respect to
> data, so you get data/mc mismatch for the energy reconstruction.
>
​Yes, of course, you would.
But the factor is applied after the trace integration - I meant it is
irrelevant if the same is done for the real data and the MC. And, as a
general rule, we must process the MC in the same way as the data in all
stages.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

-- 
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2018-03-22:
> The most consistent way would be to take the actual pulse shape, and calculate these values on the fly. 

Can you tell me what is the definition of  CALIB_SCALE? E.g. how is it computed from the pulse?  It sounds very similar to the *integration correction* we already apply, which corrects for the fact that we only integrate part of the pulse.

kosack, 2018-03-22:
In general, I agree it makes no difference, but if we want to be able to compare values with MARS/EventDisplay (particuarly cleaning thresholds, etc), it is useful to have everyone using the same unit definitions.

watsonjj, 2018-03-22:
Integration correction and calib_scale are two different correction factors. The former corrects for the part of the pulse shape thats not within the integration window, the latter corrects for the different definitions of what "1 photoelectron" is. 

In the lab we calibrate to photoelectrons by looking at the spectrum with low illumination, so we see the single pe peaks, and use the x value of the 1pe peak in that spectrum. Instead in the MC (I think this is true) pe is defined from the mean of the SPE spectrum input into simtelrray. So to convert between the two you somehow need access to the SPE spectrum that was input to create the simulation and/or create the SPE spectrum from the low pe waveforms of the MC so that you calibrate them in a similar way lab data is calibrated.

Another technique is by looking at the bias in the charge resolution, and removing it by altering calib_scale, but in my opinion that suggests there are no other factors that could be creating the bias, I think the SPE spectrum method is more correct, and is typically how one (or atleast how we, in GCT) calibrates lab data into photoelectrons.

moralejo, 2018-03-22:
On 22 March 2018 at 22:46, Jason Watson <notifications@github.com> wrote:

> ​al​
> In the lab we calibrate to photoelectrons by looking at the spectrum with
> low illumination, so we see the single pe peaks, and use the x value of the
> 1pe peak in that spectrum.
>
> ​In that process the x value is obtained by integrating the pulses in some
way, right? May be the same later used in the analysis, or just taking the
pulse amplitude.​ I think the distribution of those values is then fitted
to a sum of a noise component + 1 p.e. contribution + 2 p.e. contribution
+....  The x-distribution (resulting from the fit) for the 1 p.e. component
should be the SPE spectrum that is used in the MC simulation.

> ​​
> Instead in the MC (I think this is true) pe is defined from the mean of
> the SPE spectrum input into simtelrray.
>
> You mean then that the "1 p.e." from the lab is the x value at the *peak*
of the SPE, whereas in the MC it is the mean of the SPE, right?  If that is
the case, it is not clear to me  that we need for the two definitions.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662 <931%2064%2016%2062>
Fax: +34 935811938 <935%2081%2019%2038>
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

-- 
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

watsonjj, 2018-03-22:
> ​In that process the x value is obtained by integrating the pulses in some
> way, right? May be the same later used in the analysis, or just taking the
> pulse amplitude.​ I think the distribution of [...]

Correct. However I think you have to normalise this spectrum (with one of Konrads scripts) before directly inputting it into the simulation.

> You mean then that the "1 p.e." from the lab is the x value at the *peak*
> of the SPE, whereas in the MC it is the mean of the SPE, right?  If that is
> the case, it is not clear to me  that we need for the two definitions.

Yes I think thats how its defined

kosack, 2018-03-23:
> defined from the mean of the SPE spectrum input into simtelrray

Ah, that was the part I was missing.

dneise, 2018-03-23:
> I guess I could also add measurement_date and comment fields. That may be good for all lookups

I think a proper reference, to the paper (also internal and thus private CTA documents are fine) is needed here. 

Any 'magic' number which was the result of either a lab measurement or a simulation must have some kind of document explaining how the result was achieved and when and by whom.

dneise, 2018-03-23:
>  Or how should we compute it?

I think what can be computed from more fundamental information should be computed. This computation has also documentary value.

maxnoe, 2022-07-12:
The default for calib_scale is now 1.0, i.e. no scale, so we no longer need a reference for any value != 1.0, at least not in ctapipe itself