cta-observatory/ctapipe #678: How to reference ctapipes hillas parameter calculation in thesis?

t-fi, 2018-03-14:
I need to calculate the hillas parameters in a experimental analysis framework as part of a masters thesis. The routines in ctapipe would be very handy, since they allow arbitrary camera geometries and are conveniently written in python. The thesis itself is not directly connected to CTA.

The routines in question are implemented in:
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/hillas.py#L37
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/hillas.py#L162
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/hillas.py#L317
https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/hillas.py#L488

How do I properly reference the work? Is it enough to include BSD-3?

And as a bonus: Which of the four implementations would you suggest to use?

kosack, 2018-03-14:
Since we don't have a publication yet, I'd say you can reference CTA Observatory and link to this github site (not sure if you need to mention the license, but it can't hurt).  You should also reference one of the papers that the algorithms are based on probably, which are listed in the docstrings.

MaxNoe, 2018-03-15:
We should get a doi for this repo: https://guides.github.com/activities/citable-code/

dneise, 2018-03-15:
> Which of the four implementations would you suggest to use?

I assume this might answer your question. At least I interpreted these lines as a recommendation by the original author which version to use...

https://github.com/cta-observatory/ctapipe/blob/39cf9af6df8ffed682982aa40a0581d722db76a3/ctapipe/image/hillas.py#L661-L662

MaxNoe, 2019-11-25:
Done

closed 2019-11-25 by MaxNoe