cta-observatory/ctapipe #679: EventSource seems to iterate again after loop has finished

kosack, 2018-03-15:
Running the following code produces an unexpected result:

```py3
source = event_source(filename, max_events=2)

for event in source:
    print(event.r0.event_id)

print("after loop:", event.r0.event_id)
```

output:
```
2105
2108
after loop: 2305
```
So the event after the loop over only 2 events is done, seems to be the 3rd event. This is certainly a bug, or at least not expected behavior for an iterable.

MaxNoe, 2018-03-16:
Is that true for all EventSources or only for e.g. hessio?

calispac, 2018-03-16:
I think the reason is because of the if inside the for loop: 

https://github.com/cta-observatory/ctapipe/blob/eeb1c6bd9ae949cf4d2c2c5b389076bbd0129653/ctapipe/io/eventsource.py#L191

calispac, 2018-03-16:
Nope sorry I went to fast...

dneise, 2018-03-16:
> Is that true for all EventSources or only for e.g. hessio?

I wrote 3 tests for the 3 event sources I know about (not sure if there are more already...)
https://travis-ci.org/cta-observatory/ctapipe/jobs/354194359#L1469-L1487

(tick shows passing tests)
 - [ ] Hessio
 - [ ] TargetIO
 - [x] FitsIO (i.e. actually only SST1M ...)

I assume at the moment, this might be connected to the design decision, to modify containers in place instead of emitting fresh ones. This decision can lead to little trip-wires... you see, when the event-source gives up ownership of an event after having it returned, there is virtually no chance to modify it after the loop. 

In the implementation of the sst1m_event_source I did deliberately break this design decision (at cost of runtime).  There is a fresh Container for every event. 
https://github.com/cta-observatory/ctapipe/blob/eeb1c6bd9ae949cf4d2c2c5b389076bbd0129653/ctapipe/io/sst1meventsource.py#L30

Of course one can fix this issue ... (I did not yet search for the exact source of the problem ... ) but I think it would be helpful to walk on a simple path and enjoy your life instead of walking on a path full of land mines, and just learn how to avoid them...

dneise, 2018-03-16:
I'm just trying this ... https://github.com/cta-observatory/ctapipe/pull/680/commits/2cc44ec23fd79a57269a961ede8b9bdc8bb2719e

Will  try your proposal afterwards .. then we can see what we like better ... just letting the tests finish... 10 minutes or so

dneise, 2018-03-16:
I personally do not like such + 1  or  -1 in these places ... they are soo easy to mess up.

MaxNoe, 2018-03-16:
Yeah. I think you are missing a `yield` in your branch

dneise, 2018-03-16:
> Yeah. I think you are missing a yield in your branch
2cc44ec -> Test are passing ... 

now going to try @MaxNoe s proposal (locally tests are passing also for that one)
commit is:  6f6b973

MaxNoe, 2018-03-16:
> hm .. but: 2cc44ec -> Tests are passing ...

Are testing only event.count? Or that `list(event_source)` has len(max_events)?

MaxNoe, 2018-03-16:
Ah, sorry. Didn't realize you switched the order of the `yield` and the `if`. I like your solution better! Less code.

dneise, 2018-03-16:
Hm ... I really have a problem with this Codacy thingy ... In #680 There is a red-x next to 2cc44ec, which was fixing the bug ... just because Codacy says the constant N_EVENTS should be written in lower case letters ..

MaxNoe, 2018-03-16:
pep8 says constants should be uppercase with underscores

dneise, 2018-03-16:
Yes but apparently Codacy cannot guess if this is a constant or a variable or so ... I cannot know what my intention is, so it assumes it is a variable or so?

dneise, 2018-03-16:
Sory I was hijacking this thread... sorry .. let's remove this Codacy discussion from here. .. okay?

dneise, 2018-03-16:
Okay so I will revert to my solution and tidy up that horribly messed up PR ..

dneise, 2018-03-16:
Fixed by d3c957e4116349472f170b7003b4f0db97874fe9 in #681.

dneise, 2018-03-17:
@kosack  Thanks for merging so quickly ...  I think this can be merged.

closed 2018-03-17 by kosack