cta-observatory/ctapipe #682: test suite for all event sources

dneise, 2018-03-16:
I think it would be good to write a suite of tests for all event sources and then just apply it to all of them.
Should remove some duplication and maybe motivate event source authors to actually write tests... 

I don't know if there is a technique for this ... so could somebody nudge me in the right direction? I'd like to learn how to do this ..

kosack, 2018-03-16:
I think you can do this with parameterized tests, like we do with things like the `GeometryConverter`.  You can define a test with a special decorator that contains all the classes you want to test on for example, and it will run that test with each class.  See https://docs.pytest.org/en/latest/parametrize.html  (and also the test code for `GeometryConverter`)

kosack, 2018-03-16:
You can also separate the part that tests the resulting output into a function (not called `test_x`), that can then be called by each camera test.

dneise, 2018-03-17:
Thanks .. I'll dig a bit into that docu ... always nice to improve my pytest-foo :+1:

MaxNoe, 2020-03-20:
Several event sources got removed from ctapipe and the remaining ones are tested against each other.

closed 2020-03-20 by MaxNoe