cta-observatory/ctapipe #684: add units to waveforms and images?

kosack, 2018-03-17:
Right now we have `waveform` and `image` properties at various data levels, that are each in different units.  Would it be useful to add some units for them to avoid confusion?  It would also help to remember when a value is per-nanosecond or some integrated quantity.  I think we still have a bug where we never multiply by the sampling frequency during charge extraction, so this would make it explicit:

```py3
   from astropy import units as u

   pe = u.def_unit('photo-electrons')
   dc = u.def_unit("digital-counts") # or just use u.count
   sampling_frequency = 1 * u.GHz

   r0_waveform = (waveform * dc) * sampling_frequency
   r1_waveform = (waveform * pe) * sampling_frequency
```

It would add some overhead, of course, but possibly help avoid some errors. Of course one problem is that what we call "photo-electrons" is not really the correct term  after pedestal-subtraction and flat-fielding.

mackaiver, 2018-04-06:
👍 for using units to avoid errors in some places . While the overhead can be quite substantial in my experience, i'd rather have it correct than fast.

mackaiver, 2018-12-11:
No. But lets think about a integration/validation test. Also rename the variables so its clearer

closed 2018-12-11 by mackaiver