cta-observatory/ctapipe #685: do we need to add flat-fielding correction for MC analysis?

kosack, 2018-03-17:
In the MC analysis, does anyone know if the gain factor we get from `libhessio` is just the PE/DC ratio (absolute pixel calibration), or a combination of that plus the flat-fielding coefficient (relative pixel calibration)?  If the former, we still need to multiply the FF coefficients before calculating hillas parameters and image cleaning to get better results.

mackaiver, 2018-12-11:
We are not sure. @moralejo

moralejo, 2018-12-11:
We do not apply any additional flatfield coefficients in the MARS-based
analysis of simtelarray output. As far as I can see,thatt is also the case
in simtelarray's internal analysis. Either the flatfielding is incorporated
in the provided factors, or all pixels' gains are the same in the
simulation, rendering it unnecessary.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

watsonjj, 2018-12-11:
I believe there is a gain variation between pixels in the simtelarray simulation, and it is corrected for in the dc_to_pe factor.

moralejo, 2018-12-11:
> I believe there is a gain variation between pixels in the simtelarray
> simulation, and it is corrected for in the dc_to_pe factor.
The gain variation must indeed be corrected, because it goes in the
pixel-wise pe to ADC counts factor. But simtelarray also simulates a
gaussian spread of the pixel-wise photodetector efficiencies , which is of
3.5% (std dev) for the LST. So in principle one could achieve a more
uniform response by applying a relative calibration of the pixels. I do not
know if simtelarray's calibration factors already include such flatfielding
(hence the correction would be not ADC to p.e. in the specific pixel, but
rather "ADC to average p.e."). This is a question for Konrad, I'll ask him.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

MaxNoe, 2020-03-20:
@moralejo did you receive an answer?

FrancaCassol, 2020-03-23:
Hi,
in sim_telarray there is a "flatfielding" card that if set to 1 applies the flat-field equalisation to the coefficients. This card is set to 1 by default. I think this is the case for prod3 (can somebody else confirm this?)

moralejo, 2020-03-23:
> 1) The gain_variation parameter with
>     a) flatfielding = 0: a signal amplitude r.m.s. (common to all
>        channels/gains on a photosensor), in addition to the
>        QE variation.
>     b) flatfielding = 1: after signal amplitudes are adjusted to
>        perfectly balance any QE variation (uniform flatfield),
>        gain_variation corresponds to the r.m.s. in flatfield
>        response you get (from perfectly flat illumination).
>     Where things started to be a bit fuzzy in my memory was if
>     there is also ADC sensitivity variation involved (2% assumed
>     from compiled-in defaults are in effect). After reviewing my
>     code for initialising the electronics parameters, I decided
>     to change the logic a bit. Instead of a perfect flatfield
>     having constant QE * gain, it should rather have constant
>     QE * gain * sensitivity.
>
> *The usual calibration coefficients    reported in the simtel files always
> included complete    flatfielding* - but there is a second set of
> coefficients
>     in the data without flatfielding. So the answer depends on
>     which set got applied - or if you applied your own calibration.
>     Assuming flatfielding=1, qe_variation=0.035, gain_variation=0.03,
>     and fadc_var_sensitivity=0.02, you should have got a
>     sqrt(0.03^2+0.02^2) = 0.036 r.m.s. in relative flatfield signals
>     and would in future simulations only get 0.03.
>     With flatfielding=0, it would be sqrt(0.035^2+0.03^2+0.02^2),
>     both the old and the new way.
> @moralejo <https://github.com/moralejo> did you receive an answer?

Konrad wrote a mail on this (copied below), following discussions in a
recent LST camera MC model update call.

If I got it right, the "usual" set of conversion factors, which I think are
those we use in the various analyses of the CTA MC, include complete
flatfielding (I put the relevant sentence in bold below).  In that case, no
additional flatfielding would be needed to process the MC.

A.

wrote:

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

kosack, 2020-03-23:
ok so it sounds like it's included in the multiplicative calibration factor already, so no need to do more.  I'll close this then

closed 2020-03-23 by kosack