cta-observatory/ctapipe #687: HDF5TableWriter/Reader improvements

calispac, 2018-03-20:
Hi all,

I am still playing around the `HDF5Table` classes. So far I have experienced some issues: 

1. Appending mode is not supported. 
```python 
with HDF5TableWriter(filename, group_name='group_1') as h5_table:
    h5_table.write('data'', datacontainer_1)
with HDF5TableWriter(filename, group_name='group_2') as h5_table:
    h5_table.write('data'', datacontainer_2)
```
AFAIK this will: 
* open the table
* write `datacontainer_1` in `/group_1/data/`
* close the table
* open the table and ERASE everything
*  write `datacontainer_2` in `/group_2/data/`

I think @MaxNoe pointed that out in a recent PR

2. Writing container class is not supported: 

```python 
class ContainerA(Container):
    b = Field(ContainerB , 'a ContainerB field')
class ContainerB(Container):
   c = Field(int, 'a int field')

a = ContainerA()
with HDF5TableWriter(filename, group_name='group_1') as h5_table:
    h5_table.write('data'', a)
```
The former will not write `a` because a contains a field of type `Container` which not supported.

For the moment I think it is fine because one can do : 

```
with HDF5TableWriter(filename, group_name='group_1') as h5_table:
    h5_table.write('data'', a.b)
```

Could you confirm me that it is actually the case or am I just missing something ?
Concerning 1. I will try to play with the reader and set `mode='a'` in the `open()` function.
Concerning 2. I am not sure one can do this easily but maybe it is just a matter of adding  : 

```python
 if isinstance(value, Container):
       table_name = value.__class__.__name__
       self._create_hdf5_table_schema(table_name, value)
```

in the for loop of https://github.com/cta-observatory/ctapipe/blob/e4d9a307ec7bf2846bc5930de9b175ee9f5cf41a/ctapipe/io/hdf5tableio.py#L83-L152

kosack, 2018-03-21:
appending:
------------

Have you tried passing the `mode="a"` option when constructing it? Underneath HDF5TableWriter just passes all kwargs to `pytables.open()`, so in principle it should work, though I have not tried it. 

Writing container classes
--------------------------

I intentionally did not implement recursion of data structures, since I wasn't sure exactly how one would want that to come out: should the sub-Containers be flattened and write out a single 2D table, or should HDF5's hierarchical tables be used? (the latter breaking compatibility with Pandas, etc). So if you want to write out recursive tables, you have to set up a wrapper for that currently, or do each one explicitly.  That's actually why in the first commit, this was calle "SimpleHDF5TableWriter".

However, thanks to @MaxNoe's recent additions, you can write multiple containers at once to a table: eg. `writer.write([a,b])` in your example above.

kosack, 2018-03-21:
Also, in your example above you create 2 `HDFTableWriters` in the same file. This is not needed - the same HDFTableWriter can write to any number of tables within the same file - just call `write(table1, containers)` and `write("table2", containers)`  It handles multiple tables internally

kosack, 2018-03-21:
I think the one design decision that I should improve is that it only allows tables to be under a single sub-folder (dataset) in the HDF5 file, so if you want to put some tables in one and others elsewhere, it doesn't work.  There's no particular reason for this, other than it was easier to implement, but it would be a simple change to have the user give the full table path rather than just the name. Is that why you wanted to create more than one writer rather than just passing around a single writer  @calispac ?

kosack, 2018-03-21:
see also the DL1 dumper PR: #554

calispac, 2018-03-22:
Thanks for the reply. I think I totally missed the **kwargs thing and from previous discussions I had in mind that appending is not supported... 
So the append mode is supported but the only issue I had was that I could not append to a existing group.

closed 2019-08-05 by kosack