cta-observatory/ctapipe #691: optional dependencies - not really an option?

dneise, 2018-03-23:
I think we ran into a problem in PR #688, we cannot solve, since we cannot reproduce the problem on our machines.

There was a rather heated discussion about the introduction of telescope specific event sources a few weeks ago. The result was to introduce them as optional dependencies. I always felt this is overly complicated. We all are friends, and I have absolutely no problem with ctapipe installing dependencies on my machine, which I might never need. So in particular ... let's say I never wanted to read a TargetIO file, I have absolutely no problem if there is some targetio dependency automatically installed on my machine  in the process of installing ctapipe. And I hope other people do not mind installing a protozfits library, even if they might never need it.

In #688 we have now the case, where test is failing, which is unrelated to the code modified in that PR and I have virtually no chance to even start digging into that problem, since I do not have installed the necessary dependencies, so the test which is failing on travis is skipped on my machine.

In issue #675 "importorskip considered harmful?" I already mentioned, that skipping tests due to missing optional dependencies can lead to important tests being skipped. And now I would go even a step further and propose that optional dependencies should maybe be avoided.

I personally think that not everything that can be done also has to be done. 

An inflation of features in a project leads to a drastic decrease of developers time to maintain those features. I think ctapipe ultimately is not about supporting optional dependencies, is it? It is about doing cta low level analysis. So why do we make our lives harder they are anyway by supporting features we do not need?

dneise, 2018-03-23:
[image]

watsonjj, 2018-03-23:
If you read my comment in #688 you will see that this issue could be debugged and solved without installing the extra dependencies. You just needed to print what file path was causing the error in NectarCAMEventSource.is_compatible.

Notice the line 
```
ctapipe/io/nectarcameventsource.py:45: in is_compatible
h = fits.open(file_path)[2].header
```

Full error message (from TravisCI):

```
ctapipe/core/factory.py:245: in produce
    instance = factory._instance
ctapipe/core/factory.py:194: in _instance
    product = self._product
ctapipe/core/factory.py:171: in _product
    product_name = self._get_product_name()
ctapipe/io/eventsourcefactory.py:64: in _get_product_name
    if subclass.is_compatible(self.input_url):
ctapipe/io/nectarcameventsource.py:45: in is_compatible
    h = fits.open(file_path)[2].header
../../../miniconda/envs/cta-dev/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py:317: in __getitem__
    raise e
../../../miniconda/envs/cta-dev/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py:313: in __getitem__
    self._positive_index_of(key))
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
self = [<astropy.io.fits.hdu.image.PrimaryHDU object at 0x7f54636b5f60>, <astropy.io.fits.hdu.table.BinTableHDU object at 0x7f54636b09e8>]
func = <built-in method __getitem__ of HDUList object at 0x7f5463541e08>
args = (2,), kwargs = {}
    def _try_while_unread_hdus(self, func, *args, **kwargs):
        """
            Attempt an operation that accesses an HDU by index/name
            that can fail if not all HDUs have been read yet.  Keep
            reading HDUs until the operation succeeds or there are no
            more HDUs to read.
            """
    
        while True:
            try:
>               return func(*args, **kwargs)
E               IndexError: list index out of range
../../../miniconda/envs/cta-dev/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py:1044: IndexError
```

watsonjj, 2018-03-23:
The TARGET software libraries, svn and redmine are private. I do not have the power nor the will to change that. If someone specific requests access to the software we can arrange the access.