cta-observatory/ctapipe #693: HDF5TableReader issue with MetaContainer

calispac, 2018-03-25:
Hi all,

I came up with an issue with the reader. Actually it was a mistake from my side but I wanted to share it here. Maybe there is something to improve to avoid user mistakes.

My code was something like this : 
```python
class MyContairer(Container):
    a = Field(np.int, "a field")

... # save MyContainer using `HDF5TableWriter`

with HDF5TableReader(filename, '/data') as h5
      events = h5.read('table_1', MyContainer)
      for event in events: 
           pass
```

My mistake here was to pass `MyContainer` to `read()` instead of `MyContainer()` the error was at this line :
https://github.com/cta-observatory/ctapipe/blob/d8c8826fa33fae740d6f17ddda4e2c28e2b09fae/ctapipe/io/hdf5tableio.py#L361

It was telling me that I cannot assign to a `MetaContainer`. Maybe it is just matter of adding a few comment to the read function to tell the user to not forget about `()`.

kosack, 2018-03-25:
That's the same anywhere in python, and is not specific to this case - in this case `read()` takes a `Container`, not a `class` (which is what you passed in the first time).  `type(MyContainer)` is a class, whereas `MyContainer()` constructs an instance of the class `MyContainer`.  

By the way, it will be faster to construct a `MyClass()` , and just pass in the instance of it each time, rather than re-constructing it on each call. 

```py3
cont = MyContainer()
with HDF5TableReader(filename, '/data') as h5
      events = h5.read('table_1', cont)
      for event in events: 
           pass
```

kosack, 2018-03-25:
I'll update the documentation, since I notice it does not specify the type of the inputs to `read()`, which is probably the cause of the confusion.

commit 8bdada64 (2018-03-26): Added better documentation to TableReader.read(), fixes #693 (#694)

* Added better documentation to TableReader.read(), fixes #693

* make sure codacy knows this is python3 (was complaining about syntax)
closed 2018-03-26 by kosack