cta-observatory/ctapipe #695: Image truncation or leakage parameter missing?

rlopezcoto, 2018-03-28:
Hi,

I've been trying to find how truncated images are handled but could not find any parameter that is containing that information after the shower image analysis. Does anybody know where is this implemented or wether there is no implementation at all?

Thanks

kosack, 2018-03-28:
Right now it has not yet been merged into the main ctapipe code -there are only some basic mathods in @tino-michael's repo, including edge-pixel rejection. If you'd like to help, its appreciated! It's not clear to me yet what method is best.  The easiest of course is just putting a cut on the hillas centroid distance.

rlopezcoto, 2018-03-29:
Sorry, I could not find anything in any of the branches of the repo, @tino-michael could you point me to the modified files?

tino-michael, 2018-04-01:
It's not a branch in my ctapipe fork, it's in my `ImageCleaning` module in a separate repo:
https://github.com/tino-michael/tino_cta/blob/master/tino_cta/ImageCleaning.py

MaxNoe, 2019-01-03:
Got implemented in #783, function is `ctapipe.image.leakage`

closed 2019-01-03 by MaxNoe