cta-observatory/ctapipe #696: Recent Pull Requests were "Squashed and Merged"

MaxNoe, 2018-04-03:
The recent history has no merge commits and whole PRs were squashed into single commits.

Was this by intention? I guess you clicked "Squash and merge" by accident @kosack?

I think for a project like this, having the merge commits for PRs is mostly preferable.

kosack, 2018-04-05:
No, I intentionally have been squashing them - it makes it much easier to see what each commit was for, since mostly you get a bunch of small fixes rather than a comlete set, and most people are bad at making a clean commit history.  Before, I had a history that was basically spaghetti, and it was impossible to tell which features were added when. 

Can you tell me why having all the intervening commits are better? I don't really see it. I think that works well when you have expert git users, but not when most of the commits are fixing typos and formatting.

kosack, 2018-04-05:
Comparison:

[image]

vs

[image]

Though I agree that perhaps for very large PRs, keeping some of the history is useful.

MaxNoe, 2018-04-05:
I agree that "Fix typo in comment" commits are not very helpful.

Also 'merge branch master into <feature_branch>'. Probably for most small Pull Requests squashing is fine.

But If squashing creates giant commits, it gets harder to debug using tools like `git bisect` or revert or cherry-pick changes

MaxNoe, 2018-04-05:
Maybe we should put more focus on getting PRs to have a good history than squashing terrible ones. 

Like rebasing PRs to have a decent number of meaningfull commits and then merge.

kosack, 2018-04-10:
Yes, I agree- I'd rather have PRs with cleaner history (squashing typo commits, etc), though that requires some training, or perhaps better developer documentation. I'm just as bad at it as others, so I can't really complain too much.  

Perhaps we could add a better guide of how to do it in the docs? I'm not enough of an expert to do that , however.

dneise, 2018-04-13:
Such a guide ... or even a little hand-on training would also help me along quite well... I sometimes commit way way way too often with ridiculous messages (maybe not in ctapipe PRs but in general). 

I sometimes like to open a PR prematurely and keep working on it and committing and pushing in pseudo-real time while chatting about the PR in a telegram chat or so. If my peer is far away this create some kind of peer-programming-experience. However the resulting history is a mess.

So some hands-on tutorial about how to tidy up such a mess would help me personally a lot.

closed 2018-04-25 by kosack