cta-observatory/ctapipe #697: Possible Directional Reconstruction Biases.

mackaiver, 2018-04-06:
Hello,

I used `ctapipe.reco.HillasReconstructor` to reconstruct the directions of incoming showers. It seems to work nicely for gammas. I have the feeling the directional reconstruction for protons is biased however. While we don't really need to reconstruct proton directions *correctly* I think they should at least be reconstructed *uniformly* if you will. Look at the plot below for some comparison. It looks fine for gammas as you can see. However I would expect a much more uniform distribution for the protons.
Can anyone reproduce this issue? Maybe I'm doing things wrong.

[image]

mackaiver, 2018-04-06:
After a discussion with @TarekHC here is some more detail. 

1. The `ctapipe.reco.HillasReconstructor` class only takes the Hillas features as an input to reconstruct the direction. So there is no *training* on gammas or anything like that.

2. Both plots show the same number of points. It looks like there are less points on the right but they are simply plotted on top of each other.

mackaiver, 2018-04-06:
BTW the x axis is in degrees for both plots

vuillaut, 2018-04-06:
@mackaiver what cleaning did you use before doing Hillas reconstruction?
I suppose you work with prod3b files. In this production, the pointing direction corresponds to the gamma source direction, meaning that the source is right in the center of the field of view.
As a result, even with no cleaning and a random Hillas reconstruction, the average will give you (0,0) and you will believe you have made a good reconstruction. I do not know if that's what happened there but there is clearly a difference in the reconstruction because you are dealing with diffuse vs point source particles (rather than gamma vs protons).
For the direction there is no training, it is a simple geometric reconstruction at play.

TarekHC, 2018-04-06:
Hi @mackaiver @vuillaut 

>  I suppose you work with prod3b files. In this production, the pointing direction corresponds to the gamma source direction, meaning that the source is right in the center of the field of view.

I'm not sure what do you mean with this. Can you explain further? The source position for point gammas should be the center of the FoV while for protons it should be a 10 (? not sure) deg cone. This is what @mackaiver showed in the first left plot, which seems reasonable. The unreasonable part is that the reconstructed direction for protons is completely clustered in the center of the FoV. If the same number of events is shown in left and right, it means that there is no "cut" selecting the proton events falling close to the center of the FoV.

So my guess is that either @mackaiver has a simple bug somewhere (for example, assuming a variable means something while it really means something different), or that the direction reconstruction algorithm is wrong. Is it perhaps assuming that the source position is always in the center of the FoV?

mackaiver, 2018-04-06:
> I suppose you work with prod3b files

I do indeed. 
I use the default settings for both the calibration and the cleaning.  
```
   calibrator = CameraCalibrator(
       event_source=event_source,
   )
   for event in event_source:
       calibrator.calibrate(event)
       for telescope_id, dl1 in event.dl1.tel.items():
           mask = tailcuts_clean(camera, dl1.image[0], min_number_picture_neighbors=2)
           hillas_params = hillas_parameters(
                     camera[mask],
                     dl1.image[0, mask],
                     container=True
           )
```
If I remember correctly this used to work fine for prod3 Should I change anything for prod3b? Or did anything change in ctapipe in between?

> As a result, even with no cleaning and a random Hillas reconstruction, the average will give you (0,0)

Yes the average will be (0, 0) I agree. However if the hillas parameters are random then the direction should be random to.

vuillaut, 2018-04-06:
@TarekHC 
> Can you explain further?    

Pointing is (70, 0), Source position is (70, 0)
In H.E.S.S. we do not point directly at the source but position slightly off. For example for a source at (70, 0), you would point at (71, 0).

> or that the direction reconstruction algorithm is wrong. Is it perhaps assuming that the source position is always in the center of the FoV?   

There is no assumption there. The direction reconstruction is done by crossing lines whose direction is given by the ellipse major axis.

@mackaiver    
> Yes the average will be (0, 0) I agree. However if the hillas parameters are random then the direction should be random to.    

Actually not. The relevant Hillas parameter here is the ellipse direction. If directions are random, with several telescopes, you will statistically get (0, 0) as an average reconstruction (see my poor drawing skills), i.e. your pointing direction, and in this case (un)fortunately the source direction.
I learnt that the hard way trying to optimise the cleaning: it turned out the best cleaning was no cleaning at all.

[image]

However, let me stress that I do not know if this is the right explanation for your problem there. I am just saying that you might have a bad reconstruction for protons **and** for gammas but you cannot actually see it in this situation.

dneise, 2018-04-06:
I'm not sure I follow this argument 

> If directions are random, with several telescopes, you will statistically get (0, 0)

While the directions are random from event to event, they are not random when several telescopes look at the same shower from multiple points. So yes, the center of mass of all red points should be at (0,0), but still this argument does not explain why the red points are not scattered much wider, does it?

kosack, 2018-04-06:
> I use the default settings for both the calibration and the cleaning.

those are almost certainly not correct. You need to have the right cleaning thresholds for the given camera, otherwise you'll get crazy hillas parameters.  Did you check some images to see if they look like they are reasonably clean? 

Also note that the reconstruction does not necessarily give the "true" position for protons, except in cases where there are no sub-showers and the shower is very gamma-like. However, your plot does seem very strange. 

> So for now I assume that I have a bug. It would have come up before I suppose.

I wouldn't assume that!  it could be a bug introduced in the reco code, though I dont' see much that has changed. With limited manpower, this could be something that was overlooked (though I'm sure it did not look so bad in the past).

vuillaut, 2018-04-06:
@dneise 
> While the directions are random from event to event, they are not random when several telescopes look at the same shower from multiple points. So yes, the centre of mass of all red points should be at (0,0), but still this argument does not explain why the red points are not scattered much wider, does it?   

I am not talking about the event direction but about the ellipse direction. If the cleaning is very bad, whatever is the event incoming direction, the ellipses directions will be random. With n ellipses directions taken randomly, the reconstruction gives a reconstructed event direction at (0, 0), no matter the event. Therefore, all your events are reconstructed at (0, 0) - exactly what we see in the first plot from @mackaiver.

*NB: (0, 0) here is the position relative to the pointing direction (i.e. centre of the field of view). In absolute in the example above, it is equal to (70, 0).*

dneise, 2018-04-06:
> If the cleaning is very bad, 
AH I missed that point entirely. Thanks

mackaiver, 2018-04-06:
@vuillaut and @kosack I think you are correct. The Calibration and cleaning settings seem to be wrong. 

> Also note that the reconstruction does not necessarily give the "true" position for protons, except in cases where there are no sub-showers and the shower is very gamma-like.

Yes. And I don't need the true position. But I was expecting rather random/uniform one. 

> those are almost certainly not correct. You need to have the right cleaning thresholds for the given camera, otherwise you'll get crazy hillas parameters

That would certainly explain a lot. I just presumed this calibration factory thingy kind of 'normalizes' the pixel values for all cameras so the cleaning thresholds would all be the same.

> Did you check some images to see if they look like they are reasonably clean?

I just checked. The cleaning does not seem to work very well. It just selects all pixels. 

So all I would need to know now are the correct settings for the calibration and cleaning methods. And Also how to decide whether to use the low gain or the high gain channel. Are the correct settings  for prod3b known or even written down somewhere?

kosack, 2018-04-06:
> Are the correct settings for prod3b known or even written down somewhere?

Just grepping through some of the pipeline steering code, I think these are what are use (I may be wrong, but they don't look too far off).  Normally using something like "picture=10, boundary=5" should work for most cameras (other than FlashCam, which we don't yet calibrate correctly) though it may be a bit of a  harsh cut. 

```py3
              {"ASTRICam": (5, 7),  # (5, 10)?
                 "FlashCam": (12, 15),
                 "LSTCam": (5, 10),  # ?? (3, 6) for Abelardo...
                 # ASWG Zeuthen talk by Abelardo Moralejo:
                 "NectarCam": (4, 8),
                 # "FlashCam": (4, 8),  # there is some scaling missing?
                 "DigiCam": (3, 6),
                 "CHEC": (2, 4),
                 "SCTCam": (1.5, 3)}
```

mackaiver, 2018-04-06:
> Just grepping through some of the pipeline steering code, I think these are what are use (I may be wrong, but they don't look too far off).

Awesome. Thanks for that. 🥇 

Turns out I was using the correct cleaning all along. The calibration was wrong. 
Things look much better. See below.

[image]

I now use `r1_product='HESSIOR1Calibrator' ` which seems to do the trick. So stupid mistake on my side. But maybe also an unfortunate default setting in the calibrator.

Thanks for your input everybody.

closed 2018-04-06 by mackaiver

moralejo, 2018-04-06:
The cleaning values should be valid (not necessarily optimal, but not too
bad) if the integrated pulses are scaled by the calibration factors
provided by sim_telarray, i.e.   event.mc.tel[tel_id].dc_to_pe

Perhaps that factor has been omitted?  As we discussed recently in another
thread, there should be another factor (but close to 1) to convert it to
units of "single p.e. peak", if we decide to use those.

-- 

Abelardo Moralejo Olaizola
Institut de Física d'Altes Energies
Tel : +34 931641662
Fax: +34 935811938
Avís - Aviso - Legal Notice - (LOPD) - http://legal.ifae.es

watsonjj, 2018-04-06:
> Perhaps that factor has been omitted?

That factor is included in the HESSIOR1Calibrator

watsonjj, 2018-04-06:
Thats very confusing. That functionality is tested for in ctapipe.calib.camera.tests.test_calibrator:

```python
def test_eventsource_r1():
    dataset = get_dataset("gamma_test.simtel.gz")
    eventsource = HESSIOEventSource(input_url=dataset)
    calibrator = CameraCalibrator(eventsource=eventsource)
    assert isinstance(calibrator.r1, HESSIOR1Calibrator)
```

mackaiver, 2018-04-06:
Ah. Then maybe it is because I'm not using `HESSIOEventSource`. Here is my code 
```

    event_source = EventSourceFactory.produce(
        input_url=input_file,
    )

    calibrator = CameraCalibrator(
        event_source=event_source,
        r1_product='HESSIOR1Calibrator',
    )
```

watsonjj, 2018-04-06:
AH! you specified "event_source", is there somewhere else (an example?) with that mistake that you copied from?

mackaiver, 2018-04-06:
> is there somewhere else (an example?) with that mistake that you copied from?

Yes. This came from some discussion on the whole 'tools' and 'factory' business. I think there was also an issue somewhere where we discussed this. Can't find it right now. But his PR uses it for example #661. So we could go ahead and blame @MaxNoe for the whole thing if you like :stuck_out_tongue_winking_eye:

kosack, 2018-04-10:
> AH! you specified "event_source" instead of "eventsource", is there somewhere else (an example?) with that mistake that you copied from?

That's something we should fix... certainly it's a bug if it accepts an incorrect parameter name (or at least it's not expected behavior).