cta-observatory/ctapipe #702: Different integration windows per camera type

kpfrang, 2018-04-10:
Hi,

is there a way in `CameraCalibrator` to specify different widths and shifts for the trace integration window per camera type?

To my understanding, the DL1 container is filled for all telescopes with the values specified in `WindowIntegrator`. That implies that for all telescopes the same values will be used.

watsonjj, 2018-04-10:
There are a few ways it can be done. If you wish to configure via the command line or a config file, then you can have a look at examples/calibration_pipeline.py (line 145).

If you wish to set the values directly in the script then one way to do this is to define the individual calibration classes yourself instead of using the `CameraCalibrator` convenience class. You can find an example of this being done in examples/display_integrator.py. For example:

```python
        kwargs = dict(config=self.config, tool=self)

        eventsource = EventSourceFactory.produce(**kwargs)

        self.extractor = ChargeExtractorFactory.produce(
           **kwargs, window_width=12, window_shift=4
        )

        self.r1 = CameraR1CalibratorFactory.produce(
            eventsource=eventsource, **kwargs
        )

        self.dl0 = CameraDL0Reducer(**kwargs)

        self.dl1 = CameraDL1Calibrator(extractor=self.extractor, **kwargs)
```

watsonjj, 2018-04-10:
However, to properly address you question, applying a different charge_extraction per telescope is not something that has been implemented yet... and needs a discussion on how to best implement it...

Perhaps the extractor argument to `CameraDL1Calibrator` could either be a `ChargeExtractor`, or a list of `ChargeExtractors`, where it is assumed each index in the list corresponds to the next telescope

kpfrang, 2018-04-10:
Okay, thanks for the clarification.

> Perhaps the extractor argument to CameraDL1Calibrator could either be a ChargeExtractor, or a list of ChargeExtractors, where it is assumed each index in the list corresponds to the next telescope

Something like this was exactly what I wondered whether it already exists. I guess it would be nice to have something like this as it would make it easier to play around with the numbers.

MaxNoe, 2018-04-10:
> Perhaps the extractor argument to CameraDL1Calibrator could either be a ChargeExtractor, or a list of ChargeExtractors, where it is assumed each index in the list corresponds to the next telescop

Maybe `dict[camera-type] -> ChargeExtractor` would be the most useful

watsonjj, 2018-04-10:
> Maybe dict[camera-type] -> ChargeExtractor would be the most useful

Yes some look-up regarding regarding the default extractor per camera-type would be nice. That is one of the points I would like to discuss as outlined in #703

kosack, 2018-04-11:
In real data of course, there will be only 1 camera per file, so it's only an issue for simulation data that are merged. The main question is how to configure this - we could have a yaml or ecsv table that lists the parameters per telescope (or camera type).  Or we can use the config system, but then we need a configurable object per camera type.  It's not clear which is simplest.

MaxNoe, 2018-04-11:
> In real data of course, there will be only 1 camera per file, so it's only an issue for simulation data that are merged. 

I think this will also be a problem for real data, as you do not want to maintain 7 different analysis script and remember to call the right on for the telescope in a file.

So a single executable that can do the anlysis for every possible standard input file  would be prefereble imho.

watsonjj, 2018-04-11:
> So a single executable that can do the anlysis for every possible standard input file would be prefereble imho

Absolutely, any solution we come up with should keep this functionality.

MaxNoe, 2019-11-25:
This should be doable now with the `TelescopeParameter`, right?

watsonjj, 2019-11-25:
Added ability to set per telescope in #1155

closed 2019-11-25 by watsonjj