cta-observatory/ctapipe #703: Different telescopes and camera calibration

watsonjj, 2018-04-10:
Current state of camera calibration:
At each calibration step there is a loop over telescopes. The same calibration is applied to each telescope.

As raised in #702, it will be almost always required that each telescope type uses its own extractor. This is currently not supported in the current way we do things, and it also raised a few questions in my mind that I need help addressing:

1. Are the DL0 files still planned to be stored per telescope?
2. Currently the event containers have `tel` sub-containers. Are these planned to be kept?
3. We need some way of defining the defaults for certain Components (e.g. the default `ChargeExtractor` to use) based on the telescope. Perhaps some sort of "set-up" Component that supplies the user with the correct classes based on the telescope.
4. Should the `ctapipe.calib.camera` Components be restructured to only apply to a single telescope at a time. This would make a lot more sense in the pipeline flow. As the name suggests, these components are camera specific, and their configuration will change depending on the telescope being investigated. A wrapper class (`CameraCalibratorGroup`) could be created which loops through each telescope, and applies the provided calibrators for that telescope.

Thinking about this has made me wish that instead of the `tel` sub-container, we instead had telescope at a higher level, and beneath it are the "event" containers for each telescope. I wonder what the plan is for the future, and what are the proposed approaches to combine the data together for each telescope when the telescopes aren't all in a single file like they are for the sim_telarray output.

watsonjj, 2019-11-25:
The ability to set the extractor traits per telescope was added in #1155

The ability to choose a different charge extraction method per telescope is still unavailable (briefly discussed at Bologna meeting, #1148, won't be implemented for now)

The loop over telescopes in CameraCalibrator is still be discussed in #1165, so for now this issue will remain open.

maxnoe, 2021-05-05:
Implemented in #1661

closed 2021-05-05 by maxnoe