cta-observatory/ctapipe #705: Docs version unknown

cdeil, 2018-04-12:
@kosack - Why is your docs version unknown?

[image]

kosack, 2018-04-12:
that's strange!  I'm not sure - it used to work correctly. Hopefully it will start working again when the next tagged release is made (hopefully this week)

kosack, 2018-04-12:
No, that was removed years ago, and it's worked fine since then.  We replaced it with a custom system that detects and sets the version based on the latest git tag.  On local copies the version gets set correctly, so it's something to do with the travis build. Not much has changed there, but it could be an updated dependency broke something. This was certainly working on the automatic docs as of a month ago. 

for example, i get:

```
ctapipe-info --version

*** ctapipe version info ***

version: 0.5.2.post509+git02986f2
```

And it works when I build the sphinx docs myself. 

[image]

STSpencer, 2018-05-18:
Is this related to some of the travis CI tests failing? The most recent pull requests seem to be failing with the error:
The command "travis-sphinx -v --outdir=docbuild build --source=docs/" exited with 1.

thomasgas, 2019-02-25:
The docs displays the version nicely, so this is not an issue anymore.

closed 2019-02-25 by thomasgas