cta-observatory/ctapipe #708: Issues with ctapy Demo

Sojourner92, 2018-04-17:
The ctapy.py demo in examples/brainstorm/misc_demos is flagging up a lot of issues in codacy. I've fixed most of its complaints but need the 'events.fits' file to test it. Does anyone have it, or should we just delete the example?

kosack, 2018-04-18:
I think the events.fits file is any DL3 event file, so it should work with the data challenge data. 

However, the brainstorm code is unused even if sometimes still interesting.  At the  very least it shouldn't be checked for style. Either we drop it all, or exclude it from what pep8/pyflakes/flake8 check (in setup.cfg) and that will clean up many issues.

kosack, 2018-04-18:
I'd say probably just remove the whole brainstorm directory, and I can always add that stuff to another repo.

commit 37b605ed (2018-04-19): Remove brainstorm folder (#712)

* Deleted Brainstorm folder (deprecated)
* fixes #708
closed 2018-04-19 by kosack