cta-observatory/ctapipe #718: Code of conduct?

dneise, 2018-04-24:
Do we want to have a code of conduct?

Github thankfully provides (in my opinion) some good thoughts about how to create a welcoming community, maybe one is already aware of this. In case not, maybe one finds this interesting to read.

<https://help.github.com/articles/adding-a-code-of-conduct-to-your-project/>

We might be too small to enforce a code of conduct, but maybe we can loosely agree on one and try to follow it.

In my opinion having and following a code of conduct is similar to having and following a [Style Guide for Python Code](https://www.python.org/dev/peps/pep-0008/). For example, instead of having endless discussions about how to format code, we can simply have bots delivering bad news to us like: "Your code does not comply to our Style Guide". I do not think we want bots telling us how to behave in a discussion, but I believe it can save time, if there is an argument, to just point to the code of conduct and say: "Hey, I believe this was inappropriate."

But still ... having a code of conduct means potentially having to enforce it, which will cost time none of us might have...

kosack, 2018-04-25:
I think this is perhaps more of a question for the CTAO administration, particularly the interim software manager. It's not ctapipe-specific, but should be something we have globally for all CTA software projects. 

However, if we adopt one, it probably wouldn't be hard to convince them to simply go with what we choose.

kosack, 2018-04-25:
I've sent a mail to the software group at CTAO mentioning that this should be added to the general software management plan.

closed 2018-04-25 by kosack