cta-observatory/ctapipe #728: HESSIOEventSource reads all telescopes, even if allowed_tels is set

kosack, 2018-04-27:
The current behavior seems to be broken compared to how it was originally implemented - only the telescopes in the allowed_tels list should be filled into the event data structure, which also speeds up the IO for reading data with super-arrays. 

To check, we should add a unit test to ensure that `set(allowed_tels) == set(event.r0.tels.keys())`

MaxNoe, 2020-03-20:
We now use the eventio bases event source.
I think we should not put too much effort into fixing the hessio eventsource.

closed 2020-03-20 by MaxNoe

watsonjj, 2020-03-20:
Does the eventio package completely skip telescopes not specified in allowed_tels? And does not waste time reading them?

MaxNoe, 2020-03-20:
Yes, but the speed up is not as large as you might expect, at least in the case for compressed files. As it still has to decompress the bytes.