cta-observatory/ctapipe #738: Proposal: allow for selectable sets of datasets/configuration

kosack, 2018-05-03:
Right now we store all defaults and configuration tables in the `ctapipe-extra/ctapipe_resources` package, and the user can override them by setting the `$CTAPIPE_SVC_PATH` environment variable (which takes a directory or list of directories where to look for service files).  

This is fine for simple use cases, but is getting cumbersome when we switch simulation versions, or even later when we have multiple sites/configs.  I propose that we think of developing a more flexible system, which can allow sub-sets to be enabled by the user using a single name describing the full set of data. 

The simplest realization is to start to structure the `ctapipe_resources` directories, and allow for these different data versions (note in CTA parlance, these are *SVC* (service) data).  Perhaps:

```
ctapipe_resources/
    prod3b/
    prod4/
    ...
```

In each sub-dir, we could further organize things to make it more clear:

```
prod3b/
     cameras/
          NectarCam.camgeom.fits.gz
          FlashCam.camgeom.fits.gz
          ...
          gain_channel_thresholds.ecsv
          optics.ecsv.txt
    atmosphere/
          paranal.atmprof.fits.gz
          atmprof26.ecsv.txt
    data/
          gamma_test.simtel.gz
          gamma_test_large.simtel.gz
```

Then we provide a method to set the current data sub-set `ctapipe.set_svc_data("prod3b")` or something like that if they want to switch from the default. From an API standpoint, not much would change, since `utils.get_dataset_path()` and related functions would remain the same, just automatically look in whichever one is specified, and they already have support for searching subdirectories. 

Users could still specify custom paths using `$CTAPIPE_SVC_PATH`, but those should point to paths with sub-dirs in them with new names (e.g `my_custom_svc_data/`, which would be enabled by `ctapipe.set_svc_data("my_custom_svc_data")`.

Any better ideas?

kosack, 2018-05-03:
Note: this concept is not so different from how the IRFs are being organized for the science tools (currently using a *caldb* structure, but that's a bit cumbersome)

thomasarmstrong, 2018-05-21:
Hi Karl, 

With the configurations, would it make any sense to connect with the yaml files currently being constructed for the MC? Although I guess the problem might be that parameters are stored in non-ideal formats...
 https://forge.in2p3.fr/projects/cta_analysis-and-simulations/repository/show/Simulations/MCModelDescription/trunk/configReports

STSpencer, 2018-07-02:
Would it be possible to have the FACT open data release integrated into this? There aren't many 'real' datasets that are publicly available and that would be a useful thing to have for MC/simulation comparison purposes.

kosack, 2019-03-04:
A comment on this proposal: we should also look at what is done in *gammapy*,  where they use an index table on a server to point to datasets, which makes it easy to move the data to other servers, etc, without changing the code.

kosack, 2022-10-27:
Should use the [reference data sets](https://redmine.cta-observatory.org/projects/cta_analysis-and-simulations/wiki/Reference_data_sets) to generate the data