cta-observatory/ctapipe #739: Move some notebooks to the docs

kosack, 2018-05-03:
Since sphinx can both execute and display Jupyter notebooks (via https://nbsphinx.readthedocs.io/), it might be nice to select some of our notebook examples to include in the main docs (or write some better ones). That way they both appear in the documentation, and are executed during the TravisCI build, so we can check for errors (similar to what is described manually in #724)

kosack, 2018-05-03:
A nice example of this can be found in the [HESS GPS documentation](http://docs.gammapy.org/dev/notebooks/hgps.html)

STSpencer, 2018-09-24:
One thing that would be useful is links to examples from function references, i.e. like [matplotlib](https://matplotlib.org/api/_as_gen/matplotlib.pyplot.subplots_adjust.html#matplotlib.pyplot.subplots_adjust)

kosack, 2018-09-25:
> One thing that would be useful is links to examples from function references, i.e. like matplotlib

I'm not sure if that's possible with the Jupyter Notebook extension to Sphinx, but perhaps there is some option to link the code? I haven't tried it out yet.

MaxNoe, 2020-03-20:
THis was done some time ago.

closed 2020-03-20 by MaxNoe