cta-observatory/ctapipe #740: Ring width dependent on the optical efficiency of the system

rlopezcoto, 2018-05-09:
Brought up during a presentation in the last ASWG meeting (https://indico.cta-observatory.org/event/1686/contributions/15776/), there was an issue with the optical efficiency-dependence of the muon ring width. One of the proposed causes of this dependence was the fact that muons rings were fitted after applying image cleaning, which could get rid of some tails in the fits. I implemented the option of performing the analysis without applying the cleaning #736 and run the analysis again (it takes way longer than before so a crosscheck would be appreciated). Here are the results (top applying cleaning, bottom without):

[image]
[image]

Meaning that without applying image cleaning we still get an optical efficiency-dependence on the muon ring width. My bet is that somehow the likelihood fit performed in the likelihood `fit_muon` in `muon_integrator.py` is not doing the job completely right, maybe also related to #647. Anybody had a look to that issue?

maxnoe, 2022-12-21:
> Meaning that without applying image cleaning we still get an optical efficiency-dependence on the muon ring width. 

Actually, what you see here is probably the ring width being dependent on the ring radius, right?  And this relationship is different for different efficiencies.

I suspect this line in the muon tool is at fault:

https://github.com/cta-observatory/ctapipe/blob/a633e0fb7d0620266c77d6fd68cf085314976b42/ctapipe/tools/muon_reconstruction.py#L164

The mask around the ring which pixels enter the fit is dependent on the fitted ring radius. This is probably not a good idea and might lead to this effect.

moralejo, 2022-12-21:
I agree there is no obvious reason to use a maximum distance to the ring (along the radius) proportional to the radius itself. 

However, the fact that smaller-R rings have larger fitted widths is (I think) due both to larger ring fit parameter uncertainties for smaller rings (fewer pixels, less light) and also due to physics (fig 22 of https://arxiv.org/pdf/1907.04375.pdf). In any case, I think the 0.4 R value is pretty wide, I would not expect it affects much the result of the fitted width.