cta-observatory/ctapipe #752: Pointing Positions in EventContainer

mackaiver, 2018-06-05:
Hello,

The pointing position for each telescope are always 0 in the latest master.
Is this expected behaviour? 
```

>>>event.mc.tel[1].azimuth_raw 
0
>>> event.mc.tel[1].altitude_raw 
0
>>> event.pointing.as_dict()
{}
```

kosack, 2018-06-05:
When I try it I get:
```py3
print(event.mc.tel[34].altitude_raw)
1.2217304706573486
print(event.mc.tel[34].azimuth_raw)
0.0
print(event.mcheader.run_array_direction)
<Quantity [ 0.        , 1.22173047] rad>
```

The event-wise ones are:
```
In [15]: event.mc.alt
Out[15]: <Angle 1.1591475009918213 rad>

In [16]: event.mc.az
Out[16]: <Angle 6.153833866119385 rad>
```

Obviously this needs to be cleaned up (I think the alt/az values in the mc tel container are the header values, not the event-wise ones, which makes little sense), but it seems to be relatively ok.

kosack, 2018-06-05:
Are you sure you aren't looking at an MC that is simulated at `az,alt=0,0`? In that case, those values would never change.

kosack, 2018-06-05:
relates to: #466 #499

mackaiver, 2018-06-06:
> Are you sure you aren't looking at an MC that is simulated at az,alt=0,0? In that case, those values would never change.

Pretty sure. I tried on three different files:

```
gamma_20deg_0deg_run11194___cta-prod3-merged_desert-2150m-Paranal-3HB89-NGFD_cone10.simtel.gz
proton_20deg_0deg_run20640___cta-prod3-merged_desert-2150m-Paranal-3HB89-NGFD.simtel.gz
gamma_20deg_0deg_run993___cta-prod3-merged_desert-2150m-Paranal-3HB89-NGFD.simtel.gz
```
The output is the same for each of the files:
```
print(event.mc.tel[34].altitude_raw)
0

print(event.mc.tel[34].azimuth_raw)
0

print(event.mcheader.run_array_direction)
[ 0.          1.22173047] rad

```

I'm on the latest master with both `ctapipe` and `pyhessio`

> The event-wise ones are:
> 
> In [15]: event.mc.alt
> Out[15]: <Angle 1.1591475009918213 rad>
> 
> In [16]: event.mc.az
> Out[16]: <Angle 6.153833866119385 rad>

This one seems to be okay for me. But this is the position of the simulated source right?

kosack, 2018-06-06:
Ah, just noticed I was not in the master branch when I tried it - I'll test again.  It does look like there were some changes to that code, so perhaps something broke (and we need a unit test somewhere probably)

kosack, 2018-06-06:
I still get values with the master - can you tell me what version of pyhessio you are using? It could be a bug in that. I'm using 2.0.1

mackaiver, 2018-06-06:
I'm also on 2.0.1

```
▶ pip freeze | grep hess
pyhessio==2.0.1
```

I don't know whats going on. I'll try on another computer later this week.

mackaiver, 2018-06-11:
I still have the same problem on another machine. Can somebody else confirm?

mackaiver, 2018-06-11:
Ah. I just noticed that the pointing information is only non zero for telescopes with data. That explains it. Is that new behaviour?

dneise, 2018-06-11:
Afraid to sound dumb here but could you tell me what exactly to do, to reproduce this?

I assume one has to download some data from somewhere and execute some script, but I seem to be unable to find this information.

mackaiver, 2018-06-11:
Sure. You need some simtel file for that. I did it like this:

```
event_source = EventSourceFactory.produce(
     input_url= 'some_simtel_file',
     max_events=10,
     product='HESSIOEventSource',
)

event = [event for event in event_source][0]
ids = event.inst.subarray.tel_ids
print([(i, event.mc.tel[i].altitude_raw) for i in ids])

```

Of course I'd simply use `next(event_source)` but that doesn't work.

mackaiver, 2018-06-11:
Max just noticed that one could use `next(iter(event_source)` as well.

closed 2018-12-11 by mackaiver