cta-observatory/ctapipe #755: What about reading pointing information, auxiliary data from real data???

labsaha, 2018-06-08:
Is there any general consensus on how the pointing information should be stored in real data and how that will be read by ctapipe? 

Most probably pointing information for **all the events** might not be there in the data. It could be one value per drive report and associated time which will be attached to the event data file. If so, then ctapipe needs to interpolate (and correct) the pointing information from the drive reports for all the events in the file. Is there any development along this direction?

kosack, 2018-06-11:
There is a preliminary implementation for HESS data underway, and some discussion via email with the LST people, but generally this is a table of coefficients (like focus distance, translation, rotation) in FITS or ascii format that is interpolated  by the event timestamp and applied to the hillas parameters (or perhaps the image). Similar to what I describe in #756  

We hadn't given much thought to it since the focus was on MCs, but it would be good to do this in common for all telescope prototypes.

maxnoe, 2020-10-26:
Duplicate of #747 ?

closed 2020-10-26 by maxnoe