cta-observatory/ctapipe #756: where to put flat field calibration coefficients?

FrancaCassol, 2018-06-11:
Hi, 
I have a question somehow related to #747. 
I would like to start to implement  flat field calibration procedures in ctapipe. These procedures will produce, with time scale of around one night, several coefficients per pixel (at least one amplitude and one time but for two gains cameras also high/low gain ratio or similar ...). Where should be stored these coefficients in the current scheme? Also in the event.meta data?

Thanks for any hint
Cheers

F.

kosack, 2018-06-11:
My view so far has been that all calibrations like FF, pedestals, pointing corrections, etc should not be stored event-wise, since they are not calculated that way (they are on longer time-scales). Therefore they don't belong in the event data structure. Here is my proposal, and feel free to comment since there are other ways to do it:

1. for each calibration coefficient set (e.g. FF or pedestals), we store the coefficients in an external file (FITS o something readable by `astropy.table.Table.read()`, i.e. by `ctapipe.utils.get_table_dataset()`) that is essentially a table with at least 2 columns: *time* and *coefficient_x* (hopefully also error bars on the coefficient if we have them). If there are more coefficients, we just add columns to this table.  These are a sort of MON data (e.g. Flat-fielding monitoring, in time). 

2. we make a simple helper class (probably can derive from or be a common base class) that provides interpolation of all columns in the input file, given a time-stamp and perhaps tel_id. This should define a `__call__()` method that returns the interpolated value, so it could be swapped for other interpolators, or for any callable function that reutrns an ndarray of coefficients that is appropriate for a given camera. 

3. We pass an instance of that interpolator to the calibrator code.

For example:

```py3
flatfield = FlatFieldingInterpolator("obs0123345_flatfielding.fits")
pedestals = PedestalInterpolator("obs0123456_pedestals.fits")

calibrator = CameraCalibrator(
    eventsource=source, 
    flatfield=flatfield, 
    pedestal=pedestals, 
)
...
for event in source:
    calibrator.calibrate(event)
```

inside the calibrator, we just do something like:

```py3
# get calibration for this telescope:
flatfield_coeff, flatfield_error = flatfields(tel_id, event.dl0.time)  
pedestal, pedestal_variance = pedestals(tel_id, event.dl0.time)
pointing_correction = pointing_corrections(tel_id, event.dl0.time)
...

intensity = (image - ped) * ff_coeff) # or whatever 
```

For Monte-Carlo data, there is perhaps no interpolation to do, since there is no notion of "time", so we could just pass it a function that returns dummy values (or a single value to broadcast to all pixels):

```py3
calibrator = CameraCalibrator(
    eventsource=source, 
    flatfield=lambda t: 1.0, 
    pedestal=lambda t: 3.0, 
)
```

kosack, 2018-06-11:
I should note that FF and calibrations in my view are MON data, so indexed by timestamp as I have shown.  This proposal would actually apply to any MON or perhaps even SVC data, since it even supports the case where things don't change in time, or are loaded infrequently.

FrancaCassol, 2018-06-12:
Thanks Karl,
this is  a good starting point, I will discuss about that inside CCF and with the camera people for refining the tables content and in order to trigger first implementations

mgaug, 2018-06-15:
Hi Karl and Franca,

I agree that these data should be MON data, together with a time stamp.

The question is how large the validity time range should extend and what to do if a series of events comes to lie outside the validity range of any FF data set.

How shall the validity range be defined?

FrancaCassol, 2018-12-11:
In a certain sense, this was the starting of the discussion

mackaiver, 2018-12-11:
This is closed in light of #870. Thanks for making a PR 👍

watsonjj, 2018-12-11:
#870 Is about the generation of FF coefficients. I believe this issue is more about the application of the coefficients in the calibration.

maxnoe, 2020-10-26:
I think this is also fixed / more appropriate issues exist now. Reopen if this is a mistake.

closed 2020-10-26 by maxnoe