cta-observatory/ctapipe #768: Wierd issue with HessioEventSource

STSpencer, 2018-07-10:
Hi,
I'm having some issues using HessioEventSource. This is the code I'm trying to execute:

`    print(gamma_data) `
   ` gamma_source = HESSIOEventSource(input_path=gamma_data)`

which returns the string `/scratch/spencers/Data/Simtel/SST_GAMMAS1E5_SIMTEL.gz` followed by a `FileNotFoundError: file path does not exist:`. But the file does exist, as:
`ls /scratch/spencers/Data/Simtel/SST_GAMMAS1E5_SIMTEL.gz` at the terminal
returns`/scratch/spencers/Data/Simtel/SST_GAMMAS1E5_SIMTEL.gz`. Does anyone know what might be happening? Does the simtel file have to have a particular set of permissions for HessioEventSource to be able to read it?

STSpencer, 2018-07-10:
Found the issue, the documentation at [https://cta-observatory.github.io/ctapipe/api/ctapipe.io.EventSeeker.html#ctapipe.io.EventSeeker]() is out of date. It says the argument should be called input_path, whereas the actual argument needed is input_url.

kosack, 2018-07-12:
Yes, we should update the docs and throw a more useful error in that case, since that is somewhat confusing!

dneise, 2018-11-15:
@karl I have never used doctest before in a project, but I think doctest would have prevented this situation to occur.

https://docs.python.org/3.6/library/doctest.html

commit 9fd0d4fd (2019-01-10): Fix docstring of EventSeeker, fixes #768 (#914)