cta-observatory/ctapipe #772: Corrupted weighting in HillasReconstructor fails for small images

kpfrang, 2018-08-03:
Playing around with `HillasReconstructor`, I realized that a rather large amount of events are not reconstructed. I tracked down the problem to be the weighting of the `HillasPlane` which currently is taken as `weight=moments.intensity * (moments.length / moments.width)`.  
However, depending on the the Hillas parametrization this can lead to conflicts. For small images, where the pixels after image cleaning are all in a line, the width will be either `nan` or `0` as you can see right here:

[image] [image]

For the weighting, the means that those planes will have `nan` or `inf` as weights respectively, and those mess up the reconstruction e.g. when calculating the weighted sum of the crossings.

I think, one might still be able get some information from those images, so there might be better ways than simply throwing them away in `HillasReconstructor`. 
- there should be no `nans` as the width for correctly parametrized images but only numbers,  I guess. This would already remove the possibility of `nan` as weight
- instead of 0 one could use some very small positive number as minimum width, e.g. a fraction of the pixel size
- closely related to the last point one could put a limit on the maximum weight of the planes to avoid `inf`
- in case there is no other suited solution, one should definitely remove those images in `HillasReconstructor` so that the reconstruction can be performed with rest of images

One problem I see with the solutions would be that you somehow will set the weight 'by hand' and consequently the weights given to them might not be appropriate.  
I'm currently playing around with an other possibility of avoiding this by using different weighting methods for the planes, e.g. from a look up table.

vuillaut, 2018-08-03:
Hi
- I agree, width should not be equal to `nan` is the first case you show
- the weighting formula could be change to `intensity * (1 + length)/(1 + width)`
- in all case, one bad image giving `nan` or `inf` parameters (for any reason) should be simply thrown without throwing the event

MaxNoe, 2018-08-03:
> I agree, width should not be equal to nan is the first case you show

I disagree, `NaN` is the correct value for width in this case.

vuillaut, 2018-08-03:
> I disagree, NaN is the correct value for width in this case.
How so?
And why the event on the left would have `width = nan` and the one on the right gets `width = 0` ?
There is just a rotation between the two.

MaxNoe, 2018-08-03:
Ah sorry, of course both should be NaN or zero. They are not different at all. 

We can debate if NaN is better than zero or the other way around, but those straight line showers are clearly not useful for something that wants to weight by `length / width`

vuillaut, 2018-08-03:
Ha :)
I personally feel like a straight line is an ellipse with a `width = 0` rather than undefined width. In any case, the computation of the weights should handle this without throwing the event (nor the image) but just weighing it in a clever way. The image still gives valuable info on the source direction.

dneise, 2018-08-06:
I believe, taking into account the finite size of the bins might quite naturally avoid such singularities and hopefully would not negatively less pathological cases.

I understand the current implementation of the Hillas Parameters is the one from the [Whipple Reynolds et al 1993 paper](https://doi.org/10.1086/172269). I

In that paper the `x`  and `y` are the "phototube coordinates in degress" and it somehow assumes a phototube registers only photons coming from a very narrow direction bin, while in reality we know that the acceptance of each camera pixel is a finite range of directions. One could take this finite acceptance into account e.g. by sub-sampling the camera plane artificially say 100 times finer than it actually is and assign to this finer "grid" of positions a weight in such a way, that the total weight of all sub-samples in a pixel sums up to exactly the original charge of that pixel. 

Using exactly the same definition for the hillas parameters, just letting it act on a much finer grid, would then quite naturally avoid results like zero width. In fact the smallest width one could get then is somehow proportional of the pixel width. (In 1D it would be `sqrt(12) * bin_width`, in 2D with hex-bins, I don't know this factor by heart).

There are other techniques, but this sub-sampling technique I think is particularly simple and might already be a good start.

dneise, 2018-08-06:
Konstantin wrote:
> instead of 0 one could use some very small positive number as minimum width, e.g. a fraction of the pixel size

I think this very small positive number would indeed naturally result out of the finite acceptance and the geometry of the pixels, if using such a sub-sampling technique as I outlined above.

dneise, 2018-11-15:
Should this issue be further discussed at the meeting in december or was it solved in the meantime?

MaxNoe, 2020-03-20:
If I'm correct, this got fixed by merging https://github.com/cta-observatory/ctapipe/issues/755,
the new hillas parameters always returns a width of 0 for pixels in a straight line.

I'll add a test.

MaxNoe, 2020-03-20:
Nope, still an issue.

commit 6ff0f127 (2020-03-23): Fix hillas width 0, fixes #772 (#1240)

* Add test checking if straight lines have width 0

* Round values very close to zero to exact zero in hillas parameters

* Use a lower absolute tolerance in hillas

* Test for hillas.width in example notebook

* Check value of width

* Test exact equality, fix random seed
closed 2020-03-23 by kosack