cta-observatory/ctapipe #788: Empty HESSIOEventSource with Paranal-3HB89 files

mireianievas, 2018-09-17:
Hi,

Following some scripts from Kai, I found some Prod-3 files here: https://www.mpi-hd.mpg.de/personalhomes/bernlohr/cta-raw/Prod-3/Paranal-3HB89/ .

I have downloaded a few of them to play a little bit with ctapipe, but when I try to load  any of them following the ctapipe documentation using either

```
from ctapipe.io import EventSourceFactory
with EventSourceFactory.produce(input_url=gamma_cone.file_list[0],max_events=10,product='HESSIOEventSource') as source:
    for ev in source:
        print('test')
        break

```
or 

```
from ctapipe.io.eventsourcefactory import event_source
with event_source(input_url=gamma_cone.file_list[0]) as source:
    for ev in source:
        print('test')
```

I get nothing but an apparently empty ctapipe.io.hessioeventsource.HESSIOEventSource object. It never enters into the print('test') part. Is there anything obvious I am missing here?

Thanks,
Mireia

MaxNoe, 2018-09-18:
This works for me on the current master, having `pyhessio==2.1.0`

mireianievas, 2018-09-19:
Interesting ... I have tried with other locally avaiable files here at DESY at it works as intended

closed 2018-09-19 by mireianievas

MaxNoe, 2018-09-19:
I also tried the same file as you have