cta-observatory/ctapipe #802: timing & array plotting

thomasgas, 2018-10-16:
I'm not 100% sure, but apparently after #799 now the plotting is not working anymore in the way it used to be with #784. @LukasNickel have you checked that the example notebook in ctapipe/examples/notebooks/timing_array_plotting.ipynb is still giving good results?
I was obtaining [this image](https://github.com/cta-observatory/ctapipe/pull/784#issue-214902434) but now i get this:
[image]
Thanks!!!

LukasNickel, 2018-10-16:
I can confirm the notebook behaving like this, but i do not see any obvious mistakes right now.
I cannot recreate the bug without the ArrayDisplay aswell right now, will need a closer look.

Are we sure the arrows get flipped correctly? Im having trouble understanding which angles got flipped and why. Might be me since i dont fully understand how the angles need to be converted into the array plane. I tried figuring out which arrows got flipped by setting every slope to 1 and comparing each arrow manually which gave me the following image:
[image]

Looking at the individual slopes i would assume that telescopes 3,4,5,6 get flipped and 7,9 are not displayed. Since this doesnt happen at all and wouldnt yield the expected result either I'm probably still missing something tho.

thomasgas, 2018-10-16:
I don't exactly get what's happening here. 
If you look at the results for the gradient that I got before in the [old notebook](https://github.com/thomasgas/ctapipe/blob/69252c51cd33d39f5a84a678700183960a6137ff/examples/notebooks/timing_array_plotting.ipynb), and compare the results for the computation of the gradient with the new notebook, i get the same results, **except** for telescopes 2 and 3 (don''t consider the 6...that one shouldn't be plotted because the gradient is computed with one pixel with one value and more than 20 pixels with another value, giving a not reliable result).
So, for all the telescopes I get the same value of the gradient, **but** for 2 and 3 the **new** notebook gives:
[image]
Whereas the **old one** gives:
[image]

MaxNoe, 2018-10-17:
Is that https://github.com/cta-observatory/ctapipe/pull/782 again?

MaxNoe, 2018-10-17:
Yes it is, so let's just wait until #782 get's merged

thomasgas, 2018-10-17:
ok this is good!

thomasgas, 2018-10-17:
@MaxNoe which pixels are you setting to zero? could you past here a sample code just to understand this better?

MaxNoe, 2018-10-17:
Instead of doing

```
timing_parameters(camera[mask], image[mask], hillas)
```

do 
```
cleaned = image.copy()
cleaned[~mask] = 0.0
timing_parameters(camera, cleaned, hillas
```

This is a workaround for the bug in `CameraGeometry.__getitem__`

thomasgas, 2018-10-17:
Then I end up with something like this:
TypeError: expected x and y to have same length
in:
`slope, intercept = np.polyfit(longi, peakpos, deg=1, w=np.sqrt(image))`

I can still have a good gradient only doing this in timing_parameters:
`pix_x[~mask] = 0.0`
`pix_y[~mask] = 0.0`
`image[~mask] = 0.0`
`peakpos[~mask] = 0.0`

do you agree?

MaxNoe, 2018-10-17:
> Then I end up with something like this:
TypeError: expected x and y to have same length

This is the bug I talked about which I fixed in #804

thomasgas, 2018-10-17:
Perfect!!! I've just tested the changes and now it work with the workaround and with your fix!

closed 2018-10-30 by thomasgas