cta-observatory/ctapipe #803: Pixel group implementation

calispac, 2018-10-17:
Hi all,

After a call on calibration within ctapipe. The need to a way to access pixel groups was formulated. Pixel groups = Group of pixels that share a common hardware component. For that it was proposed to implement sparse matrices of shape = (N_pixels, N_groups). That way the average value of a calibration measurement among a group of pixels can be computed like this:

```
pedestal = np.array([?]*N_pixels)
pedestal_by_group = pedestal.dot(pixel_group_matrix) / np.sum(pixel_group_matrix, axis=0)
```
I was thinking to implement this into `CameraGeometry` does it seem to be the correct place ?

kosack, 2018-10-23:
I guess we should decide where to put these groups - do they go in CameraGeometry (which was originally intended to be only the info needed for simple reconstruction), or create a new class for camera-specific hardware details (CameraHardwareDescription? or a better name).

I guess the simplest way would be to allow groups to be defined in the CameraGeometry, perhaps by name: `geom.pixel_group['modules'],  geom.pixel_group['patches']`.  But then it is also necessary to associate those as patch_ids, and `CameraModuleDescription` or `CameraTriggerDescription` classes.  So we should think a bit about the relationships and the data model.  

Certainly I can see a few use cases like:
- sum quantities over group
- from patch group 6, see which pixels belong and from a pixel ask which group it 
- look up properties of a single element of a group (e.g. patch group 6 has which trigger threshold?)
belongs to

For the last 2, I think the sparse matrix is probably best. But then how to do the association with the group information?

kosack, 2018-10-23:
Relationships to consider in the instrument model:

- A camera *contains one or more pixels*
- A camera *contains one or more pixel groups*
- A pixel group contains *1 or more pixels* 
- A pixel group *belongs to 1 and only 1 camera*
- A pixel belongs to *1 and only 1 camera*
- A pixel belongs to *0 or more pixel groups*
- A pixel group is associated with external parameter information (E.g. a camera module group is assocated with a `CameraModuleDescription` containing details of that hardware; or a trigger patch group is associated with a `TriggerPatchDescription`),

kosack, 2018-10-23:
Perhaps we could also include neighbors as a sort of grouping, and then we can generalize neighbors and groups  to a single interface:

```py3
neighbor_matrix = geom.group['neighbors'].matrix
patch_matrix = geom.group['patches'].matrix
patch_matrix_where = geom.group['patches'].matrix_where
patch_matrix_sparse = geom.group['patches'].matrix_sparse
```

That implies some class that does all the matrix representations in common.

kosack, 2018-10-23:
This is an example from an old version of the data model, where Drawer is an example of a group (Camera Module), though that is an exclusive group (pixels belong one and only one camera module, with no overlap)

[image]

calispac, 2018-10-23:
One implementation of this we have in SST-1M is to define a class for each object Pixel, Module, Patch etc. And a Camera object being a list of Pixels, Modules etc... This seems to be the most straightforward implementation however it was difficult to ask for instance: "Does Pixel x belongs to Module y ? Is Patch x in Module y ?"

See : https://github.com/cta-sst-1m/CTS/blob/master/cts_core/camera.py

kosack, 2018-10-23:
> One implementation of this we have in SST-1M is to define a class for each object Pixel, Module, Patch etc.

We specifically avoided this for efficiency reasons - it's more useful to have the data structures unpacked, so you can operate on a pure array of pixel values, rather than a nested structure that cannot be vectorized.

kosack, 2018-10-23:
But adding a mapping of pixel -> group is quite easy and achieves the same hierarchy, i think.

kosack, 2018-10-23:
We could create a class called `PixelGroup` or 'PixelMapping'  that has input and output dimensions. 

* for neighbors, it maps `pix_id` to `pix_id`
* for patches it maps `pix_id` to `patch_id`
* for modules it maps `pix_id` to `module_id`

and provides various matrix or list representations of each mapping.  Then `CameraGeometry.group` is just a dict of PixelMappings, and these can be used in reverse to go from module_id to list of pixels in that module as well.

dneise, 2018-11-15:
Could this be implemented at the meeting in dortmund or is somebody already working on this?

MaxNoe, 2019-02-20:
Closing as a duplicate of #855

closed 2019-02-20 by MaxNoe