cta-observatory/ctapipe #805: Pixel Vertices from CameraGeometry

calispac, 2018-10-19:
Hi all,

I have some code that need to access pixel vertices. So I was wondering if there is a way in ctapipe to get the pixel vertices coordinates from `CameraGeometry`?

`CameraGeometry` is defined with:
1. pix_id
2. pix_x, pix_y (center of pixel)
3. shape (hex. or rect.)
4. cam_id
5. pix_area
6. pix_rotation
7. cam_rotation

If not I can provide a code if needed (e.g. `CameraGeometry.get_vertices()`).

MaxNoe, 2018-10-19:
```python
from ctapipe.visualization import CameraDisplay
from ctapipe.instrument.camera import CameraGeometry

c = CameraGeometry.from_name('LSTCam')
d = CameraDisplay(c)

p = d.pixels.get_paths()[0]

print(p.vertices[:6])
```

MaxNoe, 2018-10-22:
Test

calispac, 2018-10-22:
Thanks for the reply

> but this is clearly not what we want when there is no `CameraDisplay`

So should there be a method in CameraGeometry that does not use CameraDisplay in your opinion?

MaxNoe, 2018-10-22:
If you need pixel vertices for actual analysis and not visualisation, yes.

kosack, 2018-10-23:
Can you give some use case for this? If it is indeed for analysis, we can perhaps  just move the code from CameraDisplay into CameraGeometry, and that would solve it.  They may be a better representation, however (voronoi tessellation or something like that) depending on how you want to use them.

MaxNoe, 2018-10-23:
>  we can perhaps just move the code from CameraDisplay into CameraGeometry, and that would solve it. 

It's directly coming from the matplotlib `PatchCollection`, so I don't see how this can be moved from visualization to the `CameraGeometry`.

Calculating the vertices is easy and fast. Building these matplotlib patches is not.

kosack, 2018-10-23:
Ah, yes I forgot I used PatchCollection - in fact it's not hard to compute them however, and I have another implementation that computes them directly. The question is still how it will be used. If the use case is something like testing for a point within a pixel, then a quadtree or something else should be used, rather than just the points. If ifs for visualization only, it can go in the visualization class.  Usually for that is it not necessary or efficient to have the vertices for each pixel, rather to just have the verticies for a single pixel centered at 0, and use translation and rotation to place them where you need them.  For that you'd just need:

```py3
class Poly:
    """ a drawable polygon  with N vertices"""
    def __init__(self, order=6, theta0=0.0 ):
        angles = (np.linspace(0.0,360.0,order+1) + theta0) * np.pi/180.0
        self.vertices =  np.array(list(zip(np.cos(angles),np.sin(angles))))
```

calispac, 2018-10-23:
@kosack and @MaxNoe  thanks for the reply.

I should explain a bit more my use case: 

I am trying to implement pixel groups into ctapipe (see #803 ). The simplest way to implement this would be to define a sparce matrix somewhere in the code. Then I would like to visualize the camera not with pixels but with the pixel groups for cross check purposes. For this I was thinking to create a `CameraDisplay` from a `CameraGeometry` which is defined from the group vertices. For instance the code would look like:

```python
cam_geom = CameraGeometry.from_name('TELESCOPE_NAME')
pixel_vertices = cam_geom.get_vertices()
group_vertices = compute_group_vertices_from_sparse_matrix(pixel_vertices, sparse_matrix)

pedestals = np.random.uniform(cam_geom.n_pixels)
pedestals_group = pedestals.dot(sparse_matrix) / np.sum(sparse_matrix, axis=-1) 

group_camera = CameraGeometry.from_vertices(group_vertices)
group_display = CameraDisplay(group_camera)
group_display.image = pedestals_group
```

As I understood from what is described above at the moment one could do:

```python
cam_geom = CameraGeometry.from_name('TELESCOPE_NAME')
cam_display = CameraDisplay(cam_geom)
group_vertices = cam_display.pixels.get_paths()[0]

# Some code to draw the group pixels (probably trying to create a geometry from vertices) 
```

I don't know if this is more clear now. But I am wondering if `CameraGeometry` should be defined with vertices rather than how it is done at the moment. Maybe just having a `CameraGeometry.from_vertices()` would help.

calispac, 2018-10-23:
Btw I should mention that pixel groups are not necessary regular polygons

MaxNoe, 2018-10-23:
The easiest solution to visiualize these groups would be to give each pixel in a group the same value.

kosack, 2018-10-23:
We could also consider this as just an *overlay* for the CameraDisplay. I had started implementing the concept of general overlays, but never finished. Then you could overlay the group boundaries, as in the HESS camera display (Grey are the camera module boundaries):

[image]

dneise, 2018-11-15:
I think the convex hull is sometimes not what one really wants: https://github.com/cta-observatory/ctapipe/issues/600#issuecomment-357944014

MaxNoe, 2018-11-15:
Basically we want to connect the outer vertices, so the vertices that are only once or twice in the dataset, right, all others shoudl be there 3 times.

MaxNoe, 2020-03-20:
Please see #855

closed 2020-03-20 by MaxNoe