cta-observatory/ctapipe #806: `event.pointing` not filled

vuillaut, 2018-10-22:
The container `event.pointing` does not seem to be filled when reading simtel files.
I might take care of it later but I can't right now.

kosack, 2018-10-23:
Probably we need to rethink this anyhow, since events do not have pointing information, rather each telescope does (and there is no guarantee all telescopes point in the same direction). Therefore, we should only store the pointing per telescope.

vuillaut, 2018-10-23:
I see, you are right, this is certainly not where pointing should be stored.
Then maybe plainly removing this container would be better, it is a bit confusing to users I think.

kosack, 2018-10-23:
I'll close this -I added another issue to remove the pointing structure, as suggested

closed 2018-10-23 by kosack