cta-observatory/ctapipe #807: remove event.pointing

kosack, 2018-10-23:
Since there is no event pointing info, only per-telescope, we should remove `event.pointing` from the standard container structure. see #806

MaxNoe, 2019-01-03:
`event.pointing` is a map of `telescope_id` -> `TelescopePointingCointainer`, we discussed in dortmund that we want to move the pointing information there.

Maybe it should also be a container with a subfield `tel`?

kosack, 2019-01-03:
Ah yes, I think I was confusing it with the old pointing container that only contained a single position. It seems now it is closer to correct. Yes, for consistency, the Map should probably be `event.pointing.tel[tel_id]`

kosack, 2019-01-03:
I updated the title and now the task should be:

- update this to contain the pointing info for all telescopes
- fix EventIOEventSource (and HESSIOEventSource) to fill it correctly
- use it wherever pointing is needed (get rid of anywhere that uses `event.mcheader.run_array_direction`)

dneise, 2019-01-04:
Sounds rather machanical work, i.e. pretty doable. Anybody working on this?

maxnoe, 2020-10-26:
This is implemented now, I think through #1163

closed 2020-10-26 by maxnoe