cta-observatory/ctapipe #809: Question: abandon sliced images and CameraGeometries for masked variants?

kosack, 2018-10-23:
We implemented slicing for `CameraGeometry` objects to be able to unify the interface to algorithms like `ctapipe.image.hillas_parameters()`, which originaly had only `pix_x`, `pix_y`, and `signal` as arguments, but then could become `f(geom, image)` like other image algorithms. However, for speed and convenience it was useful to allow users to write `hillas_parameters(geom[mask], image[mask])` after computing the mask using `tailcuts_clean()`

However, this introduces a bit of complexity, as was shown in #782, and does not cover the use case of "broken" pixels that should be ignored or treated specially in the analysis.  

One cleaner method may be to use (optionally) masked arrays everywhere for images that have missing pixels or pixels to be ignored, which could unify  all algorithms.

To do this, one would have to do a bit more work, though the API could be improved to help with that:

```py3
mask = tailcuts_clean(geom, image)
image = np.ma.masked_array(image, mask)

params = hillas_parameters(geom, image)
```

and note that image could be designed to already be a masked array, e.g. if there were broken pixels identified in a previous step, in which case one would want to "or" together the cleaning mask with the existing mask, or for example define an algorithm for inpainting broken pixels:

```py3
image = ...  # a masked array with N broken pixels masked out
image.mask.sum() # returns N

new_image = ctapipe.image.inpaint(geom, image, method='wavelet')

new_image.mask.sum() # returns 0
clean_mask = tailcuts_clean(geom, new_image)
cleaned_image = np.ma.masked_array(image, image.mask | clean_mask)
```

note: I think masking is already used e.g. in the Target calibration

kosack, 2018-10-23:
to make this simpler, perhaps `tailcuts_clean` could simply return a masked version of image, or-ing the mask with the existing one if it exists.

kosack, 2018-10-23:
Note this would also require a modification to `CameraDisplay` to support masked arrays (right now it breaks with them)

MaxNoe, 2018-10-24:
> `assert cleaned.data is image.data # also might ensure no copy was made`

Is this wanted? I think all functions accessing image data should be pure. 
We had a lot of bad experience with functions mutating their inputs in the FACT-Tools analyis chain.

IMHO, a function that returns something should be pure, a mutating function should not return anything. That way, the API is more clear.

kosack, 2018-10-24:
> Is this wanted? I think all functions accessing image data should be pure.

I don't really mind either way - this just is what we do now, i.e that the mask and the image are stored separately and not modified.

kosack, 2018-10-24:
The one drawback seems to be that masked array operations are quite a bit slower (in my tests up to a factor of 5-10 times) than manually applying the mask and using a normal numpy operation.  That seems a bit strange to me, but I'm not sure how they implemented it (I would have assumed similar speed or faster).  So it seems to be more of a convenience than an optimization, but does make the syntax better.  

We could also define our own simple class "Image" that holds a geometry, image, and mask, and just do it all manually rather than using `numpy.ma`, then we only need to pass that to each algo, however that does add complexity.

MaxNoe, 2018-10-24:
That's not the point.

In your proposal, tailcuts now mutates it input (changing the mask of the input masked array).

Using `numpy.ma.array` you can actually create a new masked array, referencing the same data but with a different mask:

```python
import numpy as np

data = np.random.normal(size=1440)
mask = data > 1

masked = np.ma.array(data, mask=mask)
other = np.ma.array(masked.data, copy=False, mask=data < 1)

# not the same python object
print(other.data is masked.data)
# but the same memory
print(other.data.base is masked.data.base)
```

MaxNoe, 2018-10-24:
>  (in my tests up to a factor of 5-10 times)

For a simple `.sum()` it was a factor of 3, that seems hefty indeed

kosack, 2018-11-13:
it seems the current masked array implementation is in pure python, no in C as other numpy operations, which accounts for some slowness.  E.g. see first comment here: https://stackoverflow.com/questions/5760668/python-numpy-masked-arrays-are-very-slow

I guess we should perhaps just continue to keep track of the image and mask manually (or via a custom class). It's not ideal, since I like that all numpy operations work the same with unmasked or masked arrays.

dneise, 2018-11-14:
> masked array implementation is in pure python, no in C 
I am not sure this is the reason here. Looking at the implementation of [`numpy.ma.sum()`](https://github.com/numpy/numpy/blob/master/numpy/ma/core.py#L4999) and [`numpy.ma.prod()`](https://github.com/numpy/numpy/blob/master/numpy/ma/core.py#L5078) I believe one can see that these operations are not implemented as "slice then aggregate" but rather "fill-with-neutral-element then aggregate". This is needed, I believe, then the masked array is multi-dimensional, since slicing an nd-array with an nd-mask result always in a 1d-array. which is pretty useless when one would like to sum along a certain axis. That being said, I believe one could simply use `numpy.ma.array` everywhere in ctapipe and at some point implement a version which finds all places, that could be sped up in case the masked array is 1D and speed these places up. Such a "1D-accelerated version of numpy.masked.array" could be (at first) implemented in a fork of numpy and in case it works for us we could hand it in as a PR to the numpy people. I assume the if-clause needed to accelerate the 1D case is not nice, since it wastes some time, but might actually improve the performance for many people out there... so maybe the numpy people would take this PR...

For the time being however, I think, if we believe the masked array is exactly what we need, we should use it despite its apparent bad performance and speed it up later.

dneise, 2018-11-15:
I made a mistake:

>>    (in my tests up to a factor of 5-10 times)

> For a simple .sum() it was a factor of 3, that seems hefty indeed

Yes, `sum`ming is a factor of 3 slower, but simple operations like `+ - * / **` are a factor of ~10 slower. 
So the stuff I wrote above about sum and prod might be correct, but irrelevant given that simple operations are much more expensive. 

----
Side note:

Actually I tried yesterday [to speed up `numpy.ma.sum()` using `np.einsum` ](https://github.com/numpy/numpy/compare/master...dneise:masked_sum_performance) and it actually worked :-) ...
for 2D arrays the std numpy performance is like this:
```
/home/dneise/anaconda3/lib/python3.7/site-packages/numpy/__init__.py version: 1.15.1
2D - case, shape: (100, 1440)
timing: data[inverted_mask].sum().....: 458 µs
timing: masked.sum()..................: 523 µs
```

and the version using einsum is even faster than the naive implementation:
```
/home/dneise/numpy_dev/numpy/numpy/__init__.py version: 1.16.0.dev0+f783d58
2D - case, shape: (100, 1440)
timing: data[inverted_mask].sum().....: 437 µs
timing: masked.sum()..................: 154 µs
```

For 1D arrays, the version using einsum is slightly faster than the std numpy version, but still slower than the naive version: `data[~mask].sum()`.

But .. as I said ... this is all irrelevant, since I misunderstood how little the effect of optimizing `sum()` would be.

End of Side note

---

In this regard, I would like to change my opinion. I think a factor of 10 slower might actually harm the project in the eyes of some people. 

However bundling instances together in little structs or classes, which belong together, should make the code more readable and thus reduce complexity. Also it introduces a layer of abstraction, which can be used to hide the fact, that we today do not use `numpy.ma.array` but in the future when it might perform better, we could simply use it. So using something like your proposed `Image` sounds reasonable. 

However, some people might need different masks depending on context:

 * tagged as broken pixels
 * tagged as shower-associated
 * tagged as containing bright stars

So for me it is not clear, how this would look using the proposed `Image`.

MaxNoe, 2020-03-20:
I think with the performance penalties and no real benefit, we can answer this with a no.

closed 2020-03-20 by MaxNoe