cta-observatory/ctapipe #813: Tests failing with ctapipe master and latest ctapipe-extra

watsonjj, 2018-11-07:
```
ctapipe/image/tests/test_geometry_converter.py::test_convert_geometry_mock[LSTCam-001-3] FAILED
ctapipe/image/tests/test_geometry_converter.py::test_convert_geometry_mock[NectarCam-001-3] FAILED
```

Tests are failing in current master branch of ctapipe with the latest version of ctapipe-extra.

kosack, 2018-11-09:
I've identified the problem, which is a bad CameraGeometry file: `LSTCam-001.camgeom.fits.gz`.  In the code, it seems that we assume that the base name of the file ("LSTCam-001")  is identical to the header keyword "CAM_ID", since both are used in different places...

In that file, the CAM_ID is listed as "LST1Cam", while the filename is called "LSTCam-001", so some of the tests get confused by the difference. @FrancaCassol is it ok if I change the header of the file to be LSTCam-001?

```sh
% ftlist LSTCam-001.camgeom.fits.gz k
...
XTENSION= 'BINTABLE'           / binary table extension
BITPIX  =                    8 / array data type
NAXIS   =                    2 / number of array dimensions
NAXIS1  =                   32 / length of dimension 1
NAXIS2  =                 1855 / length of dimension 2
PCOUNT  =                    0 / number of group parameters
GCOUNT  =                    1 / number of groups
TFIELDS =                    4 / number of table fields
TTYPE1  = 'pix_id  '
TFORM1  = 'K       '
TTYPE2  = 'pix_x   '
TFORM2  = 'D       '
TUNIT2  = 'm       '
TTYPE3  = 'pix_y   '
TFORM3  = 'D       '
TUNIT3  = 'm       '
TTYPE4  = 'pix_area'
TFORM4  = 'D       '
TUNIT4  = 'm2      '
TAB_TYPE= 'ctapipe.instrument.CameraGeometry'
TAB_VER =                  1.0
CAM_ID  = 'LST1Cam '
PIX_ROT =                  0.0
CAM_ROT =                  0.0
PIX_TYPE= 'hexagonal'
SOURCE  = 'From real LST1 data, made by F. Cassol'
DATE    = '2019-10-02'
END
-----------------------------------------------------------------------

kosack, 2018-11-09:
I think there is also another problem with that file, specifically with the PIX_ROT value:

[image]

You can see that the pixel rotation is not correct.

kosack, 2018-11-09:
for now, I may just fix the ctapipe-extra version in environment.yml to the older one without the LST prototype camera in it until we can fix the FITS file.

kosack, 2018-11-09:
I'll move this bug to ctapipe-extra