cta-observatory/ctapipe #817: minor docu / wording question

dneise, 2018-11-15:
E.g. in the example code of Kai in  #721 we can see this line:

    mask = tailcuts_clean(...)
    [...]
    image[~mask] = 0

At least in my understanding of the word "mask" it is not immediately clear if a a `True` value in the mask refers to a pixel which survived the cleaning or not. Do you think it would make sense to use variable names like e.g. these when talking about boolean mask?

    shower_pixels = tailcuts_clean(....)
    image[~shower_pixels] = 0

Or in other contexts:
    
    broken_pixels = find_broken_pixels(...)
    image[broken_pixels] = np.nan

MaxNoe, 2018-11-15:
:+1: for better names

mackaiver, 2018-12-11:
Yes.

dneise, 2019-02-01:
This is not an action item. So it can be closed.

closed 2019-02-01 by dneise