cta-observatory/ctapipe #821: Why Factories?

MaxNoe, 2018-11-19:
At the moment, to get a correct analysis, one has to specify the product of Factories:

```python
event_source = EventSourceFactory.produce(
        input_url=input_file,
        max_events=n_events if n_events > 1 else None,
        product='HESSIOEventSource',
)
```

What then is the difference to directly using `HessioEventSource`?

watsonjj, 2018-11-19:
This surprises me... the whole point of the factory is that it finds the correct EventSource based on the input_url. The `product` traitlet exists only to force a particular return object.

watsonjj, 2018-11-19:
If the `EventSourceFactory` is not correctly returning `HESSIOEventSource` for the simtelarray file (without specifying the `product`), then the `is_compatible` function needs revisiting. Something may have changed to the simtelarray file definition....

MaxNoe, 2018-11-19:
Ok, seems to have been false alarm. Sorry.

@mackaiver always put the keys there and I remember there was an issue at least with `CameraCalibrator` doing something wrong if not provided with the products. 

On the current master, it seems to work.

kosack, 2018-11-29:
Should we close this issue, since it seems to be a misunderstanding? I thikn the use case for Factories is quite clear and reasonable (since we have now 4+ data formats to support)

kosack, 2018-11-29:
You can always use the shortcut `event_source()` helper function as well to avoid thinking about factories and products, though it's less flexible when used in a Tool since the Config doesn't propegate.

MaxNoe, 2018-11-30:
I should like another Interface.

I think it would be great if `EventSource(path)` would return the correct subclass.

watsonjj, 2018-11-30:
I considered such a functionality when designing the factories. But I found it important to separate the base class from the factory class. One reason was to avoid the subclasses of EventSource and ChargeExtractor from inheriting the factory behaviour.

Personally, I don't see what is gained over the `event_source()` helper function. But I look forward to seeing your proposed implementation.

kosack, 2018-12-03:
Hiding the Factory design pattern in the base class would be very confusing - that's not how most python classes work, and almost certainly would create circular dependencies, since the base class would need to know about all sub-classes (but sub-classes, which are plug-ins can be in different files, that require the base class to be loaded first), so I'm not sure how easy it would be to make something like that.

kosack, 2018-12-03:
In any case, this may be connected to #844

MaxNoe, 2019-02-20:
closed by #916

closed 2019-02-20 by MaxNoe