cta-observatory/ctapipe #824: Astri has no timing information?

MaxNoe, 2018-11-19:
It seams that the Astri DL1 container is missing the peakpos (it's all zeros):

```python
from ctapipe.io import EventSourceFactory
from ctapipe.utils.datasets import get_dataset_path
from ctapipe.calib import CameraCalibrator

input_url = get_dataset_path('gamma_test_large.simtel.gz')
event_source = EventSourceFactory.produce(input_url=input_url)

calibrator = CameraCalibrator(
    eventsource=event_source,
)
reco = HillasReconstructor()

for event in event_source:
    calibrator.calibrate(event)
    for telescope_id, dl1 in event.dl1.tel.items():        
        camera = event.inst.subarray.tels[telescope_id].camera
        if camera.cam_id == 'ASTRICam':
            print(np.any(dl1.peakpos > 0))

```

Is this expected? Does ASTRI not provide timing information?

kosack, 2018-11-29:
The DL1 container **does not** come from the raw data - it's all values computed from the DL0 waveforms, so it makes sense that there is no timing info, since there is no way to compute it if there are no waveforms.  I guess you mean Prod4 will have some timing info in R1/DL0?   If that is the case, it would be nice to crosscheck the timing calculation in ctapipe with that coming from the cameras/simtel (e.g. compare DL0 timing parameters to DL1 timing parameters).

kosack, 2018-11-29:
to be more clear: `dl1.peakpos` is filled  by the CameraCalibrator (somewhat confusing, i agree - we should perhaps put that elsewhere), and is not in the raw data for *any* camera.

closed 2018-12-11 by mackaiver