cta-observatory/ctapipe #828: Impact reconstruction still fails for different azimuth directions

MaxNoe, 2018-11-21:
Impact reconstruction on current master still fails dependent on pointing azimuth:

The files I used are here (standard cta password):
https://nextcloud.e5.physik.tu-dortmund.de/index.php/s/nA5SrraRjjCZRPc

Azimuth 180deg:

Azimuth 0deg:
[image]
[image]

I'm using this code to do the analysis:
```python
import astropy.units as u

from ctapipe.io import EventSourceFactory
from ctapipe.calib import CameraCalibrator

from ctapipe.image.cleaning import tailcuts_clean, number_of_islands

from ctapipe.reco import HillasReconstructor

# unoptimized cleaning levels, copied from 
# https://github.com/tudo-astroparticlephysics/cta_preprocessing
cleaning_level = {
    'LSTCam': (3.5, 7.5, 2),  # ?? (3, 6) for Abelardo...
    'FlashCam': (4, 8, 2),  # there is some scaling missing?
    'CHEC': (2, 4, 2),
}

allowed_cameras = {'LSTCam', 'CHEC', 'FlashCam'}

input_url = '/home/maxnoe/Downloads/gamma_20deg_180deg_run7360___cta-prod3-merged_desert-2150m-Paranal-3HB89-NGFD.simtel.gz'
event_source = EventSourceFactory.produce(input_url=input_url, max_events=8)

calibrator = CameraCalibrator(
    eventsource=event_source,
)

reco = HillasReconstructor()

for event in event_source:
    print(f'Id: {event.count}', f'E = {event.mc.energy:1.3f}', f'Telescopes: {len(event.r0.tel)}', sep=', ')
    
    calibrator.calibrate(event)
    
    # mapping of telescope_id to parameters for stereo reconstruction
    hillas_containers = {}
    pointing_azimuth = {}
    pointing_altitude = {}
    time_gradients = {}
    
    for telescope_id, dl1 in event.dl1.tel.items():
        camera = event.inst.subarray.tels[telescope_id].camera
        
        if camera.cam_id not in allowed_cameras:
            continue
            
        image = dl1.image[0]
        peakpos = dl1.peakpos[0]
        
        boundary, picture, min_neighbors = cleaning_level[camera.cam_id]

        clean = tailcuts_clean(
            camera, 
            image,
            boundary_thresh=boundary,
            picture_thresh=picture,
            min_number_picture_neighbors=min_neighbors
        )
        if clean.sum() < 5:
            continue
        
        hillas_c = hillas_parameters(camera[clean], image[clean])
        leakage_c = leakage(camera, image, clean)
        n_islands, island_ids = number_of_islands(camera, clean)
        
        timing_c = timing_parameters(camera[clean], image[clean], peakpos[clean], hillas_c)
        
        hillas_containers[telescope_id] = hillas_c
        
        # ssts have no timing in prod3b, so we'll use the skewness
        time_gradients[telescope_id] = timing_c.slope.value if camera.cam_id != 'CHEC' else hillas_c.skewness
        
        pointing_azimuth[telescope_id] = event.mc.tel[telescope_id].azimuth_raw * u.rad
        pointing_altitude[telescope_id] = event.mc.tel[telescope_id].altitude_raw * u.rad
    
    if len(hillas_containers) < 2:
        continue
        
    stereo = reco.predict(
        hillas_containers, event.inst, pointing_alt=pointing_altitude, pointing_az=pointing_azimuth
    )
    
    plt.figure()
    angle_offset = event.mcheader.run_array_direction[0]
    disp = ArrayDisplay(event.inst.subarray)

    disp.set_vector_hillas(hillas_containers, time_gradient=time_gradients, angle_offset=angle_offset, length=500)
    plt.scatter(event.mc.core_x, event.mc.core_y, s=200, c='k', marker='x', label='True Impact',)
    plt.scatter(stereo.core_x, stereo.core_y, s=200, c='r', marker='x', label='Estimated Impact',)

    plt.legend()
```

MaxNoe, 2018-11-21:
I think the issue is, that the pointing positions are not taken into account here:

https://github.com/cta-observatory/ctapipe/blob/5e71b43c9d11a41e09442e2f02347529a26eae85/ctapipe/reco/HillasReconstructor.py#L287-L292

mackaiver, 2018-11-21:
The result of this method is in the 'TiltedFrame' of all the cameras. It needs to be converted to the 'GroundFrame'. As far as I understand. For that you need the pointing. Maybe the conversion is simply missing?

MaxNoe, 2018-11-21:
I added a transformation from `TiltedGroundFrame` to `GroundFrame` like this:

```
        ground_frame = GroundFrame(pointing_direction=pointing_direction)
        tilted_frame = TiltedGroundFrame(pointing_direction=pointing_direction)

        core_pos_tilted = SkyCoord(
            x=core_position[0] * u.m,
            y=core_position[1] * u.m,
            frame=tilted_frame
        )

        core_pos = core_pos_tilted.transform_to(ground_frame)

        return core_pos.x, core_pos.y
```

But now the reconstruction is broken for both files.

ParsonsRD, 2018-11-21:
Hi, the transformation from the ground frame to the tilted frame is base on a projection on to the specified plane (usually based on the telescope pointing direction). Therefore it is not actually obvious how to return this information to the ground system. For example if you take the position of the telescope in the tilted system it's not actually possible to determine its original 3d position in the ground frame.

So there are 2 options how to get back the first is what you use above, which is to take the 3d position of the location in the tilted frame, but for the core location this is not actually what we want (I guess the z position here is far above the ground). Instead we want to project the tilted position back onto the ground plane. In this case you can use the project to ground function included in the coordinates module.

MaxNoe, 2018-11-21:
Ah, and the Telescope positions need to be converted into the tilted frame, right?

thomasgas, 2018-11-21:
yes! keep everything on the tilted for reconstruction and then transform back to the ground one

MaxNoe, 2018-11-21:
That is also not being done at the moment

thomasgas, 2018-11-22:
You PR fixes these plots, right? 
could you update them below here in order to visually show that the PR solves this issue? thanks

MaxNoe, 2018-11-22:
From the 0deg diffuse file:
[image]

From the 180deg diffuse file:
[image]

From the 180deg point-like file:
[image]

thomasgas, 2018-11-22:
Seems good! thanks @MaxNoe!

commit d97aed62 (2018-11-26): Fix impact reco, fixes #828, #829 (#830)

* Remove unused imports

* Transform to tilted frame and back, use psi instead of phi

* Simplify positions transformation

* Fix typo and docstring

* Add example

* Use a fixed offset instead of length