cta-observatory/ctapipe #831: Could `Tool` use `self.__class__.__name__` to create `name` instead of devs setting `self.name`?

dneise, 2018-11-22:
[Recently I stumbled over `Tool.name` again](https://github.com/cta-observatory/ctapipe/pull/827#discussion_r235436580).

So I would like to re-raise the question if `Tool` could possibly avoid introducing a new concept of classes having field called `name` which is conceptually used like `__class__.__name__`

Looking at the part in `Tool` where `self.name` is actually used: 
https://github.com/cta-observatory/ctapipe/blob/5e71b43c9d11a41e09442e2f02347529a26eae85/ctapipe/core/tool.py#L161-L182

I am wondering under what circumstances it would help, when `class MyTool(Tool)` sets `self.name = "FooBar"`? I have the impression `self.name` is usually used in logging and Provenance contexts where using the actual class names instead of some other name can only help developers and users to find out what was going on.

I also looked at some developers choices of setting `name` for their `Tool`s:

### `name` not set

Some developers do not set `name` at all. In that case `name` is automatically set to `"application"` which for me is quite surprising, c.f.:
```python
# from core/tests/test_tool.py
class MyTool(Tool): 
    pass

tool = Tool()
print(tool.name)  # --> "application" 
```

Examples for not setting `name`:
 * `MyTool` in tests
 * `class Flow(Tool)`
 * `class PipeGui(Tool)`

### `name` like class name but converted from "CamelCase"  to "ctapipe-dash-connected"

This kind of mapping of the class name to the name of the executable seems to be useful for command line tools so that users calling `ctapipe-camdemo` also see the string `"ctapipe-camdemo"` in the log output and are not surprized by seeing `"CameraDemo"` instead. However as we can see below, this kind of renaming is not always consistent and it certainly does not make a developers job easier.
 
Examples:
```
class CameraDemo(Tool):
    name = u"ctapipe-camdemo"
class DumpInstrumentTool(Tool):
    name = 'ctapipe-dump-instrument'
class DumpTriggersTool(Tool):
    name = 'ctapipe-dump-triggers'
class SingleTelEventDisplay(Tool):
    name = "ctapipe-display-single-tel"
class ImageSumDisplayerTool(Tool):
    name = "ctapipe-image-sum-display"
class MuonDisplayerTool(Tool):
    name = 'ctapipe-display-muons'
class SimpleEventWriter(Tool):
    name = 'ctapipe-simple-event-writer'
```

In this context it might be interesting to look again at the `setup.py` line which defines the entry points:

https://github.com/cta-observatory/ctapipe/blob/5e71b43c9d11a41e09442e2f02347529a26eae85/setup.py#L32

Together with:
```python
class CameraDemo(Tool):
    name = u"ctapipe-camdemo"
```

We realize `name` is an exact copy of the entry points name. 

### `name` is equal to class name

Here the `name` is identical to the class name, but manually duplicating the name, thus allowing for typos.

```
class ChargeResolutionGenerator(Tool):
    name = "ChargeResolutionGenerator"
class ChargeResolutionViewer(Tool):
    name = "ChargeResolutionViewer"
class BokehFileViewer(Tool):
    name = "BokehFileViewer"
class DisplayDL1Calib(Tool):
    name = "DisplayDL1Calib"
class DisplayIntegrator(Tool):
    name = "DisplayIntegrator"
```

MaxNoe, 2018-11-22:
I think this is just another part of all that surprising, hidden and not greatly documented behaviour of `traitlets`.

dneise, 2018-11-22:
I think Max is referring to the part "name is not set, but ends up being application". And I agree, this comes from `trailets.Application` which is a parent of `ctapipe.Tool`. 

However the point  I am trying to make is: Can we shield developers of concrete `Tools` from even knowing about this strange behavior of traitlets by setting `name` in `Tool` and not in the concrete implementations. For example like this:
```python
class Tool(Application): 
    def __init__(self, *args, **kwargs): 
        self.name = self.__class__.__name__   # or any other function of class name
        super().__init__(*args, **kwargs) 
```

dneise, 2018-11-22:
If we could do something like this neither devs not readers of a concrete implementation of `Tool` would have to bother about `self.name`

maxnoe, 2023-01-04:
We now use the entry point name as `Tool.name`, i.e. `name = "ctapipe-process"` for the `ProcessorTool`.

closed 2023-01-04 by maxnoe