cta-observatory/ctapipe #832: Could Tool use __doc__ to create description?

dneise, 2018-11-22:
This issue similar to #831.

I'd like to spare all of us the time of explaining in great detail why I believe `descripiton` might be confusing since `__doc__` already exists, but in many tools I have seen, the classes docstrings where not used to document the class, instead description was used.

I wonder, could we do this ... 
```python
class Tool(Application): 
    def __init__(self, *args, **kwargs): 
        self.name = self.__class__.__doc__ 
        super().__init__(*args, **kwargs) 
``` 
... and then devs can do the very natural ...
```python
class MyTool(Tool):
    """MyTools Description"""
    pass

tool = MyTool()
print(tool.description)   # --> "MyTools Description"
```
... and we have nicely decoupled the strange behaviour of traitlets?

kosack, 2018-11-29:
I think we even already do that in some cases (though I had used the module-level docstring rather than the class-level one), but it could be automated as you say 

https://github.com/cta-observatory/ctapipe/blob/e7152482461d1f386f3a998e6ac567e21bcb93f7/examples/display_summed_images.py#L10-L19

kosack, 2019-05-22:
this is now done in the latest version. The doc comes from either the docstring or a description attribute, whichever is available.

closed 2019-05-22 by kosack