cta-observatory/ctapipe #836: ValueError when closing empty Serializer

yrenier, 2018-11-23:
I'm using the Serializer as this:
```
from ctapipe.io.serializer import Serializer
from ctapipe.io.containers import HillasParametersContainer

output_file = Serializer('test.fits', mode='w', format='fits')
data_to_store = HillasParametersContainer()
for events in event_stream:
    output_file.add_container(data_to_store)
output_file.close()
```
That works great unless my event_stream is empty. In that case, I get an exception:
`ValueError: Inconsistent data column lengths: set()`
I would except an empty file, or no file at all, but not an exception.
That is very easy to reproduce with just:
```
from ctapipe.io.serializer import Serializer
output_file = Serializer('test.fits', mode='w', format='fits')
output_file.close()
```

MaxNoe, 2018-11-23:
This is actually astropy:

```python
from astropy.table import Table
t = Table()
t.write('test.fits')
```

dneise, 2018-11-26:
Is anybody working on this?

MaxNoe, 2020-03-20:
The current astropy version can write an empty fits table

closed 2020-03-20 by MaxNoe