cta-observatory/ctapipe #843: Decide on standardized I/O formats

kosack, 2018-11-29:
- configuration: yaml? json? a table?
- time-series or calibratiion data: FITS? HDF5? any astropy.table
  format? And what are the requirements for tables? 
- requirements:
  - keep units
  - support storage of CTA core metadata headers (provenance, etc)
  - support Pandas? (it tends to drop headers, but is nice)
  - support AstroPy.table?
  - support hierarchical tables in HDF5?
- Event output: hdf5? FITS? H5py vs PyTables vs astropy.table?

watsonjj, 2018-11-29:
I'd like to give my input here. But I am not entirely objected to any of the suggestions.

I typically use YAML for my configuration files. They are easy for a human to read and write.

I typically use HDF5 for storage of data (due to its ease of interaction with pandas). I find Pandas dataframes to be a very powerful tool for analysis.

Metadata can be stored in HDF5 with pandas:
```python
store.get_storer('data').attrs.metadata
store.get_storer('data').attrs.something_else
```
Does this fix the header problem?

I'm not sure how to best store units in HDF5, and I don't think Pandas supports them in their dataframes (https://github.com/pandas-dev/pandas/issues/10349). But one option is to create our own interface that stores the units in the `attrs` as above, and restore the units on read...

mackaiver, 2018-12-11:
So far preferences are: yaml for config, hdf5 and fits for outputs. and pytables for hdf5

closed 2018-12-11 by mackaiver