cta-observatory/ctapipe #845: have TableWriter handle Maps correctly:

kosack, 2018-11-29:
append name of map to each column of it's children
e.g. `reco['hillas1']` -> `hillas1_length`, `hillas1_width`, (see the similar functionality for adding container-wide tags)

This will make it much easier to include multiple parameterizations, or multiple image cleanings, reconstructions, etc, in the output tables.

maxnoe, 2021-05-25:
Is this still an issue now that this is essentially solved using the `DataWriter`?

kosack, 2021-05-25:
I think we can close it - the current implemnetation is to write separate tables for each item in the Map, and when reading of course the user can join them using `astopy.table.hstack` which has an option to append a prefix to common columns.

closed 2021-05-25 by kosack