cta-observatory/ctapipe #851: define how to handle things that need a configuration per-telecope in a standard way

kosack, 2018-11-29:
Many algorithms have telescope or camera-specific parameters that need to be tuned. Right now our config system doesn't handle that well, and some people have hard-coded scripts that read YAML files, tables, etc.

Let's pick a standard way. 

It should support:
* setting parameters for all cameras of the same type/version
* setting parameters for a *specific* camera by `tel_id`
* setting parameters for a group of telescopes specifid by tel_id. 
* May be useful to have some execution ordering to the configuration so one can e.g. first set defaults for all telescopes, change a few parameters for a group of telescopes(e.g. by type), and then override a few parameters for specific telescopes.

watsonjj, 2018-11-29:
> May be useful to have some execution ordering to the configuration so one can e.g. first set defaults for all telescopes, change a few parameters for a group of telescopes(e.g. by type), and then override a few parameters for specific telescopes.

I think a default configuration is correct, which might specify some things per type, and others per telescope. But I think the user should then be allowed to override the defaults in any way they wish. I.e. if the user configures all telescope types in some way, this also overrides the default "per telescope" configuration for that telescope type.

This should be very simple to implement, and most intuitive. The configurations are just applied in the order they are specified. E.g.
```python
config = {**default_config, **user_config}
```

watsonjj, 2018-11-29:
Okay, actually probably more complicated than that, as you will need some way of overwriting the "per telescope" configuration with the "per telescope type" configuration... but I think this entire topic is a step in the right direction

kosack, 2019-03-22:
see some discussion in #1022

kosack, 2019-04-02:
Just as another use-case, here is an example of what is used in Julien's pipeline for configuring 2 cleaning types (tailcuts + wavelet) as a yaml file.  
```yaml
cleaning_reco:  # note that there are more than one image cleaner defined, this is just the reco one
  tail:
   thresholds:  # picture, boundary
    - LSTCam: [6, 3]
    - NectarCam: [6, 3]
   keep_isolated_pixels: False
   min_number_picture_neighbors: 2

  wave:
   # Directory to write temporary files
   #tmp_files_directory: '/dev/shm/'
   tmp_files_directory: './'
   options:
    LSTCam:
     type_of_filtering: 'hard_filtering'
     filter_thresholds: [3, 0.2]
     last_scale_treatment: 'drop'
     kill_isolated_pixels: True
     detect_only_positive_structures: False
     clusters_threshold: 0
    NectarCam:  # TBC
     type_of_filtering: 'hard_filtering'
     filter_thresholds: [3, 0.2]
     last_scale_treatment: 'drop'
     kill_isolated_pixels: True
     detect_only_positive_structures: False
     clusters_threshold: 0

cleaning_energy:
 ...
```

maxnoe, 2021-05-05:
This is now implemented using `TelescopeComponent`

closed 2021-05-05 by maxnoe