cta-observatory/ctapipe #855: Make generalized pixel groups in CameraGeometry

kosack, 2018-11-29:
```py3
neighbor_matrix = geom.group['neighbors'].matrix
patch_matrix = geom.group['patches'].matrix
patch_matrix_where = geom.group['patches'].matrix_where
patch_matrix_sparse = geom.group['patches'].matrix_sparse
```

This implies something like a PixelMapper class that has an input and output
dimension (e.g. for neighbors, it's pixel_id -> pixel_id, where as for
patches it's pixel_id -> patch_id)

kosack, 2018-11-29:
this is a repeat of #803 , so I will close it

MaxNoe, 2019-05-28:
I found that one could use shapely to combine patches using ids and then get objects one can plot nicely with matplotlib:

```
import matplotlib.pyplot as plt
from matplotlib.patches import RegularPolygon, PathPatch
from matplotlib.collections import PatchCollection
from matplotlib.path import Path
import numpy as np
from fact.instrument.camera import get_pixel_dataframe
from shapely.geometry import MultiPolygon, Polygon as ShapelyPolygon
from shapely.ops import cascaded_union

df = get_pixel_dataframe()

fig, axs = plt.subplots(1, 3, figsize=(15, 5))

for ax in axs:
    ax.set_xlim(-200, 200)
    ax.set_ylim(-200, 200)
    ax.set_aspect(1)

pixels = PatchCollection([
    RegularPolygon(xy=(x, y), numVertices=6, radius=9.55 / np.sqrt(3))
    for x, y in zip(df['x'], df['y'])
])
verts = np.array([p.vertices for p in pixels.get_paths()])

for ax, key in zip(axs, ['CHID', 'bias_patch_id', 'trigger_patch_id']):
    ax.set_title(key)
    patches = []
    for patch_id, g in df.groupby(key):

        polygons = [ShapelyPolygon(verts[i]) for i in g['CHID']]
        boundary = cascaded_union(polygons)

        if isinstance(boundary, MultiPolygon):
            paths = [
                Path(np.array(polygon.exterior.coords))
                for polygon in boundary
            ]
            path = Path.make_compound_path(*paths)
        else:
            path = Path(boundary.exterior.coords)

        patch = PathPatch(path)
        patches.append(patch)

    col = PatchCollection(patches)
    col.set_array(np.arange(len(patches)))
    col.set_edgecolor('k')
    col.set_linewidth(1)

    ax.add_collection(col)

plt.show()
```

result for FACT's pixels, bias groups and trigger groups:

[image]