cta-observatory/ctapipe #859: Read DL1 data.

mackaiver, 2018-12-05:
Related to #556. There is the 'dl1-data-handler' project to read and write DL1 data.
https://github.com/cta-observatory/dl1-data-handler
Two things I'd like to discuss:

1. Do we want to add this functionality to ctapipe?
2. Related to #858. Do we want to organize DL2 data in the same way as the DL1 data?

vuillaut, 2018-12-05:
It seems a discussion has started within CTA to discuss a DL1 data structure and format.
@kosack is participating to this discussion.

After having used it for a while, I'd say the structure existing in the [DL1 data handler](https://github.com/cta-observatory/dl1-data-handler) is quite nice and efficient for its primary purpose (machine learning applications).

kosack, 2018-12-07:
I think it needs some revision before we define it as the official DL1 format, since there are a lot of hard-coded things still in the code, and re-definitions of some things ctapipe defines as standard. THe idea is quite good though. 

We have a second DL1 format as well (non-standard, but much simpler and pandas compatible) that is being used in the monte-carlo pipeline. In that one, no images are stored, only parameter sets for "classical" machine learning. So at some point we may need a special review meeting of intermediate data formats that support all use cases, not just convolutional learning, which isn't the highest priority.

mackaiver, 2018-12-11:
At the framework sprint we decided that we want to both read and write the format suggested in https://github.com/cta-observatory/dl1-data-handler. @nietootein @bryankim96  

> So at some point we may need a special review meeting of intermediate data formats that support all use cases

Let's set up a call. Karl will send around a doodle.

closed 2018-12-11 by mackaiver

vuillaut, 2018-12-11:
That is great news.
The writer is under revision to reach 0.7 with some functionalities 
I have started working on a reader for ctapipe but help is always very much welcome.

nietootein, 2018-12-11:
Excellent news, indeed. Looking forward to having that call.