cta-observatory/ctapipe #860: eventio library

mackaiver, 2018-12-05:
I want to discuss which eventio library do we want to use. There is `pyhessio` and the pure python library prepared by @dneise @MaxNoe  dubbed `pyeventio`

Some work is going on here: https://github.com/dneise/ctapipe/pull/2

Options are:
* Support both forever
* Choose one over the other now
* Support both and check again in a year or so.

jacquemier, 2018-12-10:
+1 If we can drop pyhessio.
As discussed with Max N, I have no problem with dropping pyhessio.

mackaiver, 2018-12-11:
We will go with pyeventio. See https://github.com/fact-project/pyeventio/

mackaiver, 2018-12-11:
@jacquemier great work with pyhessio btw. The pyeventio implementation was unittested against its output.

closed 2018-12-11 by mackaiver