cta-observatory/ctapipe #866: Remove duplicate HillasParameters implementations.

mackaiver, 2018-12-10:
As the title says. Right now we have 4 or 5 hillas implementations. I suggest to choose one of them.

vuillaut, 2018-12-10:
I approve.
From my tests, 4 and 5 never fail while 1, 2 and 3 sometimes fail for some parameters (`nan` value returned).

mackaiver, 2018-12-11:
We decided we need a test using some 'true' listed values. Also checking for all quadrants.

@MaxNoe mentioned he has some code doing that already in the toy-eventsource. 

We want to keep number implementation number 5 (cov, eigen)

vuillaut, 2018-12-11:
Hi.
How would the true values computed in the first place?

Why implementation 5?

You may find a notebook to compare the different implementation of `hillas_parameters` in ctapipe there:
https://github.com/cta-observatory/cta-benchmarks/blob/master/hillas_parameters.ipynb

The results for `skewness` and `psi` are quite different.

MaxNoe, 2018-12-11:
Because it's the most readable and runtime is not largely different (50% from slowest to fastest)

MaxNoe, 2018-12-11:
> The results for skewness and psi are quite different.

Just minus signs and rotations by 180°, so no real different, there is a unit test in ctapipe showing this.

vuillaut, 2018-12-11:
Ok thank you for the explanations @MaxNoe !

MaxNoe, 2018-12-11:
Or not rotations but different domains of the angle. So some functions return [-pi, pi], which is not really helpful, as only a 180° range is meaningfull, and some use different domains with a width of 180°.

And the sign of skewness depends on the orientation of psi

commit c5f609b8 (2019-01-04): Remove all hillas_paramters but version 5, fixes #866  (#904)

* Remove all hillas_paramters but version 5

* Add test testing hillas with toy shower

* Moar tests

* Fix test

* Remove type annotation
closed 2019-01-04 by MaxNoe